simonsobs/sotodlib #10: Does the select() method work as intended?

keskitalo, 2019-03-28:
I got an unexpected result trying to select a band from the Hardware object:

```
In [12]: hw2 = hw.select(match={'band':'UHF1'})                                                                                                                                   

In [13]: hw2.data.keys()                                                                                                                                                          
Out[13]: odict_keys(['cards', 'crates', 'bands', 'tubes', 'telescopes', 'wafers', 'detectors'])

In [14]: hw2.data['bands']                                                                                                                                                        
Out[14]: 
OrderedDict([('UHF2',
              OrderedDict([('center', 280.0),
                           ('low', 259.0),
                           ('high', 301.0),
                           ('bandpass', ''),
                           ('NET', 400.0),
                           ('fknee', 50.0),
                           ('fmin', 0.01),
                           ('alpha', 3.5)])),
             ('UHF1',
              OrderedDict([('center', 225.0),
                           ('low', 208.0),
                           ('high', 242.0),
                           ('bandpass', ''),
                           ('NET', 400.0),
                           ('fknee', 50.0),
                           ('fmin', 0.01),
                           ('alpha', 3.5)]))])

```

tskisner, 2019-04-01:
I think there is something lacking in the documentation...  Essentially this is how it works:

1. The get_example() function creates a hardware model with just a few fake example detectors.  This allows dumping it out as an example of the schema, but is not useful until one simulates some detectors and replaces the fake ones.

2.  When selecting detectors, only the **detectors** matching the criteria will end up in the output hardware model.  However, all supporting information that is referenced is kept.  So for example, a UHF1 detector is in a pixel on a UHF wafer.  A UHF wafer references 2 bands (UHF1 and UHF2), so both those bands are preserved in the output hardware model so that no references are broken.

Here is an example:
```
import pprint

from sotodlib.hardware import get_example, sim_telescope_detectors

pp = pprint.PrettyPrinter(indent=2)

# Default hardware config with a few fake detectors
hw = get_example()

# Add all LAT detectors to hardware config
dets = sim_telescope_detectors(hw, "LAT")
hw.data["detectors"] = dets

# Just the UHF1 **detectors**, plus all supporting references.
hw2 = hw.select(match={"band": "UHF1"})

# The selected dets
pp.pprint(hw2.data["detectors"].keys())

# The wafers needed to support the selected detectors...
pp.pprint(hw2.data["wafers"].keys())

# The bands needed by any wafers to support the detectors...
pp.pprint(hw2.data["bands"].keys())
```
Output is:
```
odict_keys(['00_000_UHF1_A', '00_000_UHF1_B', ... '05_429_UHF1_A', '05_429_UHF1_B'])
odict_keys(['01', '04', '03', '05', '02', '00'])
odict_keys(['UHF1', 'UHF2'])
```

keskitalo, 2019-04-01:
Thanks for the clarification!

closed 2019-04-01 by keskitalo