simonsobs/sotodlib #24: Tag a release

zonca, 2019-07-02:
I'll be using `sotodlib` for my next map domain simulation (https://github.com/simonsobs/map_based_simulations/issues/12), would it be possible to tag a release so I can more easily refer to a specific version?

tskisner, 2019-07-02:
Sure- you should have permission to do this, correct?  Probably a good idea to tag a version anyway that is compatible with the 2.2 toast before the pipeline updates for 2.3.

zonca, 2019-09-19:
what about assigning a repository maintainer that takes responsibility of tagging releases?

mhasself, 2019-09-26:
I propose we tag these reference points for now:
- cf1237123d48d90e1d5c8141b74438cdeb84eb98  v0.0.1 (Initial commit)
- 05ff8bb08d455ab1adcce10ce24151feefe26bb0  v0.1.0 ("support custom SO models for PySM 3")
- aca85843b70b0c6ebac031aa48fff47f93ed6661 v0.2.0 (merge toast_refactor)

Objections?

mhasself, 2019-09-30:
I've made a few tags... v0.2.1 is the first one where `__version__` reports the version.

closed 2019-09-30 by mhasself