simonsobs/sotodlib #32: Remove matplotlib.use

zonca, 2019-09-19:
I get some warnings when I build the docs for `mapsims` due to the line:

    matplotlib.use("pdf")

at https://github.com/simonsobs/sotodlib/blob/master/sotodlib/hardware/vis.py#L9

Would it be possible to remove that line and have the user choose the matplotlib backend before importing the library?

tskisner, 2019-09-19:
The problem is that the focalplane plotting is basically garbage when you use a raster backend.  Any of the vector backends (pdf, svg, cairo) work fine.  I chose the pdf backend since it seems to be universally available.  If we remove this line then most users will do nothing and end up with the default backend and very ugly plots that can't zoom nicely.

How is mapsims generating the docs?  The sotodlib docs seem to render fine with sphinx on readthedocs:

https://sotodlib.readthedocs.io/en/latest/cli.html#visualization

zonca, 2019-09-19:
What about switching backend inside a function?

keskitalo, 2019-09-19:
I've also seen issues from this backend being forcibly set. For instance, I cannot put a beak point in the code and use the Python debugger to plot data. This is definitely one of the instances where it is justified to conditionally import inside a function rather than at the top of the file.

tskisner, 2019-09-19:
Recall that the matplotlib backend can only be set before the import of pyplot.  It will raise an exception if you try to do it afterwards.  In this case, maybe that is a good thing.  How about this solution:

1.  move the lines:
```
import matplotlib
matplotlib.use("pdf")
import matplotlib.pyplot as plt
```
into a small "set_backend" function.

2.  call this function at the start of plot_detectors()

3.  If someone tries to call plot_detectors() **after** using other matplotlib calls, then it will raise an exception, forcing them to either manually call the "set_backend" function prior to running their other plotting or to just do the detector plotting in a separate session.

Is that an acceptable solution?

zonca, 2019-09-19:
For my needs, yes

zonca, 2020-02-05:
I was about to open another issue about this!
it would be easier just to remove this and have the user set the backend. But your solution is fine as well.

tskisner, 2020-02-05:
The motivation for setting the backend to a vector graphics one is that the detector visualization will be garbage with the default raster backend.  If we remove the backend setting completely, I am guessing that most people will just use the default matplotlib backend on their system and wonder why their plots look like crap :-)  Perhaps we could first try moving it to a function and if that still causes problems we could remove it and check the backend at plotting time and print a big obvious warning if matplotlib is using a raster backend...

zonca, 2020-02-05:
Yes a function is not a problem. Should I make a PR?

tskisner, 2020-02-05:
Sure, thanks!

PR #43 (2020-03-09): Modify matplotlib backend to PDF inside plot function
Closes #32
commit 37e95a51 (2020-03-09): Modify matplotlib backend to PDF inside plot function

Closes #32
commit f0c5fedf (2021-01-28): Modify matplotlib backend to PDF inside plot function

Closes #32
closed 2020-03-09 by tskisner