simonsobs/sotodlib #36: Travis tests fail

keskitalo, 2019-09-26:
Travis tests are failing due to missing so3g and spt3g dependencies. Furthermore, the failure does not generate non-zero exit code: [Travis log](https://travis-ci.org/simonsobs/sotodlib/jobs/589633759).

tskisner, 2019-09-26:
Thanks for opening this.  The initial sotodlib unit tests only did simple things like visualize the focalplane.  We have now expanded the package and the unit tests to include so3g and toast, so we need to restructure our travis environment.  Currently we are using the travis-provided python.  We'll need to make the following changes:

1.  apt-get a recent boost from the ubuntu-r-toolchain-test source.  Note that this **may** force us to use the python3 compatible with this boost from that apt source (I'm not sure yet).  If possible, we could continue using the travis-provided python3

2.  we should test at least an "old" and a "new" gcc version in our build matrix.

3.  Compile spt3g and so3g prior to sotodlib install / test

4.  Pip install toast from github prior to sotodlib install / test

BrianJKoopman, 2019-09-27:
I'm interested in what the solution to this ends up being. I was trying to setup the so3g repo's travis environment to build spt3g but was running into compatibility issues with boost and python3. I don't believe I was using the ubuntu-r-toolchain-test source though.

tskisner, 2019-09-27:
I'll take this one on, since we do a similar build matrix in the toast unit tests.  In that case, the compilation of dependencies with different gcc versions took too long for travis, so we pre-build those for every gcc version inside a travis docker container and then just download and untar them during travis tests.  In this case (assuming we can apt-get boost), the spt3g / so3g / toast compilation should run fine within the travis limits.

mhasself, 2020-04-16:
For OCS use, we (somehow) already build and distribute images with spt3g + so3g.  Does that make this easier?  @BrianJKoopman @tskisner

tskisner, 2020-04-16:
I was going to revisit this after doing #56 .

mhasself, 2020-04-16:
That's fine.  When the time comes, please consult with Brian and Jack as they are building official SO G3 docker images already.

tskisner, 2020-04-16:
Ok, I'm not sure one can **run** a docker container inside a travis job.  I believe the travis job is already running inside a container...

On the other hand, if we already have separate unit tests for building spt3g + so3g on the compiler versions we care about, then we don't have to test that here in our job matrix.  In that case we can get boost from apt packages, compile spt3g + so3g with whatever the default compilers are, and then pip-install toast.

BrianJKoopman, 2020-04-16:
You can run docker containers within travis. We're doing that now to run tests that need so3g. That's happening in both the so3g and ocs repos, see the test stage:
https://github.com/simonsobs/ocs/blob/master/.travis.yml

That said, Ubuntu 18.04 has since become available as a platform in Travis, which previously was a limitation in building so3g on Travis. Probably building directly on the platform could be achieved now, though using the containers allows you to avoid building the dependencies again.

tskisner, 2020-04-16:
Ok, good to know, that will make the tests run much faster!

mhasself, 2020-07-07:
Fixed in #72 !

closed 2020-07-07 by mhasself