simonsobs/sotodlib #44: Azimuth sign convention

mhasself, 2020-03-19:
In the pipe-s0001 and s0002 sims, the sign convention for azimuth is non-astronomical.  Azimuth should increase from North towards East.  This is a perennial issue... standard az, el have an opposite convention to most other standard longitude-latitude angle pairs.

For SO data files, we need to store boresight azimuth with the astronomical convention.

keskitalo, 2020-03-19:
Thanks for catching this! Indeed, I used the ISO spherical coordinate convention for Az/El in TOAST, rather than the astronomical convention. I opened a TOAST issue: https://github.com/hpc4cmb/toast/issues/326 and will try to address it shortly.

keskitalo, 2020-03-19:
Addressed in #45. This was not an issue in TOAST internal conventions.

tskisner, 2020-03-30:
Closed by #45

closed 2020-03-30 by tskisner