simonsobs/sotodlib #55: Should AxisManager restrict editing to arrays mapped with labeled axes?

kmharrington, 2020-04-15:
Just as a quick example. Let's say I wanted to sort the detector axis of `AxisManager.signal` by some value like detector angle. (azimuth or elevation offsets might be more realistic things). Right now it's possible to do this:

```
aman.array_data.array_angle = np.sort(aman.array_data.array_angle)
...
do some other things
...
# me later, thinking I could get low angle timetreams
low_angles = aman.array_data.array_angle<10
aman.signal[low_angles,:]
```
Obviously the indexing there is totally wrong. 

Possibly worse, I can do:
```
aman.signal = aman.signal[bool_mask, :]
```
without getting an error. 

I'm sure neither of these will fall under "AxisManager best practices," but should we include more checks / restrictions to prevent it from being so easy?

mhasself, 2020-04-16:
There is, in theory, a way to have the AxisManager reorder some index, such as to sort dets by some criterion.  Not sure of the detailed call right now but it amounts to making an ordered list of detector names you want and then calling _restrict()_ on the 'dets' axis.  That will return an AxisManager where all members have been re-ordered properly on the dets axis only.  I admit I'm not sure what happens if you re-order or sub-select an axis in an AxisManager that is a child of another AxisManager... that could be a disaster.

Protecting access to arrays could be done using numpy "writeable" flag.  This is quite interesting.  We could store each array in the AxisManager with writeable=False, thereby making it immutable.  Alternately we could leave writeable=True but have getattr and/or getitem return _views_ of the underlying data where writeable=False -- this would prevent a casual user from editing a buffer, while providing for the possibility of a careful user to acquire a direct reference to the underlying data.

But then, in order for someone to modify the underlying data, they would need to get that writeable reference somehow... leading to more dangling (writeable!) references in the namespace, which is potentially worse than what we have now.

The use case for modifying in place is when re-using buffers for, e.g., tod -> map -> tod, or tod -> fft -> tod.

The philosophy of AxisManager so far has been "help the user do complicated things across multiple arrays efficiently" rather than "prevent the user from doing silly things accidentally".  It's perhaps possible to do both... needs some thought though.

mhasself, 2020-07-07:
@kmharrington I guess close this issue, or continue the conversation?  I think there's nothing to be done about people storing incorrect data.  If it's any comfort, I'm pretty sure I would have seen that np.sort() in a PR and sounded an alarm... :)

kmharrington, 2020-07-08:
Yeah, sounds like a plan.

closed 2020-07-08 by kmharrington