simonsobs/sotodlib #62: tests.test_hardware takes too long

mhasself, 2020-04-29:
I think we should use smaller test cases, by default, so the tests are light enough to be run frequently.  This will encourage use of the tests.

In cases where performance "at scale" needs to be checked, individual tests can be run with additional options set; but that need not be the default.

tskisner, 2020-04-29:
The run time of those tests is basically just the plotting.  In fact there is already an override to disable the plotting:

https://github.com/simonsobs/sotodlib/blob/master/tests/test_hardware.py#L30

This is used on travis to speed things up.  We could always make that the default.

mhasself, 2020-04-29:
That would help... but it still takes ~45 seconds, with plotting disabled.

tskisner, 2020-04-29:
Ok.  I can simulate a subset of the detectors to reduce that time if we need.  When someone is developing code and running tests locally, they probably just want to run the tests for the feature they are working on (not all of them).  We should add that feature to setup.py if needed.

If we are running all the tests on travis for a PR, I think it is ok for the total to take 5-10 minutes.  However, I agree that simple tests of hardware simulating should not be using up all of our time budget.

mhasself, 2021-01-19:
@tskisner This continues to bug me... please address in the next week or so, or I can put in a fix.

tskisner, 2021-01-19:
Sure, will reduce the number of wafers in the tests.

PR #98 (2021-01-20): Reduce size of simulated hardware test.
Running the tests in this branch locally takes about 20s on my laptop:
```
python setup.py test
<snip>
Ran 39 tests in 20.255s

OK
```
If I disable plotting, I get:
```
SOTODLIB_TEST_DISABLE_PLOTS=1 python setup.py test
<snip>
Ran 39 tests in 10.738s

OK
```
Is this acceptable?  If so, then this closes #62
closed 2021-01-20 by tskisner