simonsobs/sotodlib #65: No "focalplane" in observation when loading existing data

keskitalo, 2020-05-01:
It appears that a "focalplane" entry to the observation dictionary only gets created when running `toast_so_sim.py` without existing TOD.

keskitalo, 2020-05-11:
PR #68 started as a response to this issue but so far I have been unable to reproduce the problem. @satoru99 can you share the pipeline that produces this issue?

satoru99, 2020-05-12:
As follows
`$ cd pwg-scripts/pwg-hwp/simulation/pipe-hwp-s0001/genesis
$ ./run.sh
$ python
Python 3.7.3 (default, Jun 18 2019, 11:09:40)
[GCC 7.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import toast
>>> from sotodlib.io import toast_load
>>> comm = toast.Comm()
>>> data = toast_load.load_data("datadump_SAT_MFF1", obs=["CES-Atacama-SAT1-Tier1DEC-035..-045_RA+040..+050-0-0"], comm=comm, dets=None,detranks=1)
>>> data.obs[0]['focalplane']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyError: 'focalplane'`

keskitalo, 2020-05-12:
Thanks @satoru99 . Are you working in a branch? I refreshed `pwg-scripts` and `run.sh` completes without an issue. Looking at the parameters, it does not seem to specify `--import-dir`.

If I modify `run.sh` and add `--import-dir /mnt/so1/shared/todsims/pipe-s0002/v2/datadump_SAT_MFF1` I get an error but it is not related you the focalplane:
```
TOAST INFO: Processing MC = 0
TOAST INFO: Simulating noise
TOAST INFO: Simulate noise:  77.34 seconds (1 calls)
Proc 1: Traceback (most recent call last):
Proc 1:   File "./toast_so_sim_hwp.py", line 368, in <module>
    main()
Proc 1:   File "./toast_so_sim_hwp.py", line 288, in main
    so_tools.export_TOD(args, comm, data, totalname, schedules)
Proc 1:   File "/mnt/so1/users/keskital/software/soenv/lib/python3.6/site-packages/toast/timing.py", line 43, in df
    return f(*args, **kwargs)
Proc 1:   File "/mnt/so1/users/keskital/software/soenv/lib/python3.6/site-packages/sotodlib-0.3.0+6.g88491e6-py3.6.egg/sotodlib/utils/pipeline_tools/export.py", line 60, in export_TOD
    for schedule in schedules:
Proc 1: TypeError: 'NoneType' object is not iterable

application called MPI_Abort(MPI_COMM_WORLD, 6) - process 1
```

satoru99, 2020-05-12:
run.sh is OK. The problem is in the interactive python part analyzing the simulated data output.

keskitalo, 2020-05-12:
I see. `load_data`  only loads the TOD. Some of the hardware information has to be loaded separately. Please take a look at https://github.com/simonsobs/sotodlib/blob/master/sotodlib/utils/pipeline_tools/observation.py#L182-L236

closed 2025-04-17 by mhasself