simonsobs/sotodlib #71: Store instrument parameters in HDF5

zonca, 2020-06-06:
For the map domain simulation software, I wanted to have a standard format where I can store instrument parameters, so not only Simons Observatory but also other instruments (e.g. Planck).

I designed this simple file format in HDF5 which for now only includes the subset of the parameters I am interested in, but it could be expanded to include more.

I like that it makes it easier to track versions and is just a file, independent of a software package.

In case anyone is interested, see https://nbviewer.jupyter.org/gist/zonca/ba73f601ae62b440d571b63e33623f1d

@tskisner @keskitalo @smsimon @mhasself @aiolasimone

aiolasimone, 2020-06-08:
The link is broken for me, but it does seem useful. 
@zonca, could you please bring this up at the next TOD call? so we can discuss how this compares with https://github.com/simonsobs/instrument_model

zonca, 2020-06-08:
@aiolasimone it is the same thing, but in 1 file only (binary in HDF5) instead of many

the link seems fine to me, you can try also this one (doesn't render on mobile): https://gist.github.com/zonca/ba73f601ae62b440d571b63e33623f1d

another option would be to have a repository full of many CSV files, so it is easier to make diffs, and then a script that packs them into a single HDF5 format file. (This script should run automatically after each commit, so that a versioned file is always up-to-date).

I think this is suitable for offline discussion, no need to discuss at TOD call.

aiolasimone, 2020-06-08:
Thanks for the clarification. As long as the single file and the `instrument_model` repo are in sync there should be no risk (that was the original question I meant to ask).

Your automatic HDF5 single-file packaging after each commit seems very clean and useful, I would be in favor, of course more relevant people should comment.

zonca, 2021-02-11:
no need to implement here, I will keep using my export script.

closed 2021-02-11 by zonca