simonsobs/sotodlib #83: Should so3g/spt3g need to be installed to import sotodlib.core?

mhasself, 2020-07-23:
Compiling so3g/spt3g is not straight-forward.  Many modules in sotodlib are potentially useful even in the absence of G3 stuff.  So it may be worth mandating that core is at least _importable_ without G3 stuff being available.

Currently only core.flagman is causing trouble, and that could be easily fixed by moving imports into functions.  But we should declare the general policy.

tskisner, 2020-07-23:
If we avoid "namespace" imports, then individual files in `core`, `io`, `tod_ops`, etc can depend on things like `so3g` and `toast` while still allowing imports of other files in those subpackages.  The downside is we would need to directly import `sotodlib.core.g3_core`, `sotodlib.core.toast_data`, etc rather than having those tools available by importing `core` itself.  I personally think that is fine, and it would allow grouping (for example) all the I/O stuff into the `io` package while allowing individual files in there to have different external dependencies.  Thoughts?

Of course at some point we should also work on making it easier to install spt3g / so3g...  I have some experience now building pip packages containing bundled compiled libraries, so could try to work on that at some point.  I think the main challenge with such packaging is that the compiled binaries are very large (the build directory for spt3g is 100MB on my laptop).  I think this is mainly due to having cereal generate many combinations of template types and archive types, but have not looked deeper into that yet.

kmharrington, 2020-07-24:
I'm not a big fan of moving imports into functions (I'm pretty sure that breaks PEP8). It makes it harder to track dependencies while looking through software and is a nightmare to debug when it's happening "everywhere."

I can accept avoiding the namespace imports if it's a sort of "pick your poison" sort of thing.

tskisner, 2020-07-24:
I agree @kmharrington , imports within functions break PEP8, and should only be used when there is a really good reason.  For example, if the import is very expensive and/or only needed in special circumstances.

`so3g` is part of our official software stack.  It seems like we should make that easier to install.

mhasself, 2020-10-13:
Thanks folks... I concur that ``so3g`` is too fundamental to try to keep it as a soft dependency.

closed 2020-10-13 by mhasself