simonsobs/sotodlib #92: Test failures not recognized as such in workflow

mhasself, 2020-10-21:
On my recent PR (#90), there's an "All checks have passed" report...  But in fact there's a test failure -- browse to the bottom of the "Run Serial Tests" log in:
https://github.com/simonsobs/sotodlib/pull/90/checks?check_run_id=1266734744

Expected behavior here is that a test failure like this should, at some point, produce a red X instead of a green checkmark in the PR!

(To be clear, I know why the test failed, and have a fix.  This issue is about the github workflow configuration.)

tskisner, 2020-10-21:
Hmmm, ok will look into it.

tskisner, 2020-10-21:
The `docker run` command here:

https://github.com/simonsobs/sotodlib/blob/master/.github/workflows/test.yml#L24

Runs a bash script which calls python:

https://github.com/simonsobs/sotodlib/blob/master/.github/workflows/run_tests.sh#L11

If the python command returns a non-zero exit code (I think the python unit test runner does return a non-zero code if any tests fail), then the "set -e" in the bash script should exit immediately and pass that return code back from the docker environment.  Somewhere this process is breaking down.  Will investigate more.

tskisner, 2020-10-23:
If you rebase #90 against master, hopefully you can check that this is fixed.

mhasself, 2020-10-27:
Fixed in #94.  Thanks!

closed 2020-10-27 by mhasself