simonsobs/sotodlib #105: Add requirements file

tskisner, 2021-02-03:
Hosting `sotodlib` on PyPI is a work in progress, pending so3g/spt3g work.  However, when pip installing sotodlib we could already provide a requirements.txt file that eases the installation / verification that dependencies are present.

kmharrington, 2022-01-07:
Could we revisit this? We need a requirements file for the docker install of sotodlib that goes inside of the sodetlib docker image. 

A couple notes:
* @jlashner  Does the docker image building automatically look for requirements files and install those packages? 
* Is toast ever going to be a requirement for sotodlib? I believe we decided on no but want to make sure we're still good with that. Because we definitely don't want to require toast on the smurf servers.

kmharrington, 2022-01-07:
In particular this error shows up because our filtering functions need `pyfftw` but that's not installed in the sodetilb docker. But I think adding / using the requirements file is actually the correct way to fix that problem

kmharrington, 2022-08-26:
Does this part of `setup_opts` effectively work as a requirements file? https://github.com/simonsobs/sotodlib/blob/master/setup.py#L49

tskisner, 2022-08-26:
The long story about requirements files an setup.py is here:
https://packaging.python.org/en/latest/discussions/install-requires-vs-requirements/#requirements-files
https://caremad.io/posts/2013/07/setup-vs-requirement/

But for our purposes (assuming all requirements are listed in setup.py) we could just add a `requirements.txt` file that had:
```
--index-url https://pypi.python.org/simple/

# By default, use our abstract dependencies in setup.py for the
# concrete dependencies from PyPI.
.
```