simonsobs/sotodlib #113: Correctly Handling the Brightness Temperature

Zhilei, 2021-03-16:
Following @mhasself 's suggestion, we will ONLY use T_bb (or T_CMB) for SO to avoid the T_RJ and T_CMB confusion.

The current planet simulation routine assumes a linear relationship between the brightness temperature and the spectral radiance which should be addressed in this issue.

Zhilei, 2021-03-22:
I have added three functions in the sim_sso.py script for this:
1. tb2s (blackbody temperature to spectral radiance s_nu)
2. s2tcmb (spectral radiance s_nu to t_cmb)
3. tb2tcmb (blackbody temperature to t_cmb)

I assumed the temperatures listed in the .h5 file are blackbody temperatures, so I also changed the following script to turn the blackbody temperatures into t_cmb, which can be linearly diluted.

Please let me know what you think

Zhilei, 2021-03-22:
The branch that carries the change is 'zhilei_dev', maybe there is a better way to show the changes, like a pull-request?

nad-dch, 2021-03-22:
Hey Zhilei ! Thanks for writing down these functions. There is something that confuses me. According to @mhasself's email we should be converting from spectral radiance to Kcmb units after diluting with the ratio of the planet / beam solid angle. That being said I think you should be calling instead tb2s after loading the temperature values from the .h5 file, then use the temperature to get the signal amplitude and then use the s2tcmb function (but please correct me if I am wrong).

Zhilei, 2021-03-22:
Thanks for the comment!
My understanding is that although the Planck Law does not provide a linear T_blackbody-radiance relation, the t_cmb is defined linearly with the spectral radiance, so I think we can perform the dilution on the t_cmb directly. That's probably the beauty of t_cmb!

mhasself, 2021-03-22:
Zhilei, indeed what you want to do is create a PR for this, and we can discuss more there.  Github has a "mark as draft" feature; that can be used to let people know that you want to discuss / modify the PR further before truly proposing to merge it.

This looks more or less ok.  It is indeed safe to apply the beam dilution after the conversion to a spectral radiance proxy.

Minor point ... I think "thermodynamic" should be used instead of "thermal dynamic".  But I think "blackbody brightness temperature" is a more accurate description of T_B.

Zhilei, 2021-03-22:
Thanks for the instructions, @mhasself ! I wish I knew better so that I won't bother you with all the technical details.

I have started a 'draft' pull request and I eliminated the term 'thermaldynamics' (or my poor English version 'thermal dynamics') entirely. Instead, I am just using 'blackbody' as you suggested.

Now we have conversations here on this issue and on the pull-request. How do you guys normally do this to avoid confusion from conversations happening at multiple locations?

mhasself, 2021-03-22:
You move a discussion by saying you are moving the discussion, and providing cross-links.

Let's continue discussion on the PR, #115

closed 2021-03-25 by Zhilei