simonsobs/sotodlib #119: AxisManagers -- Non-aligned Information?

kmharrington, 2021-04-16:
Looking for thoughts on how we might want to encode information in Axis managers that aren't aligned along axes. 

The main example I have here are units. We'd like a way to track units of signal. But I could also imagine we might want to encode units of other arrays pointing/encoders in particular.

My first two thoughts on options:

1. make `aman['units'] = 'string'`
2. Have some sort of `aman._units = {}` that keeps track of `{'field_name' : unit}`

jeffiuliano, 2021-04-16:
The second option there makes sense to me.  We could then set up each other object to have an eg,  `aman.signal.units`, which is like a property, that just does  `aman._units.get(name)` or `aman._units[name]=unit` under the hood (would need to look at how aman is built more closely to know if something like that is straightforward to do)

tskisner, 2021-04-16:
Consider using astropy units for labels / attributes / traits of your data.  They can be used to express even complicated compound units:

https://github.com/hpc4cmb/toast/blob/e247526822a14b9a0c7634bc9c7e02c0fcbdf2d0/src/toast/tests/instrument.py#L53-L62

Note that I'm **not** suggesting that you use `Quantity` objects for the full data (i.e. wrapped numpy arrays with a unit associated to the array), since I'm not sure of the performance of manipulating those or passing the buffer to compiled code.  However, if you have a scalar "units" label to your object, you could just use astropy units for that variable.  Or maybe a string representation of the unit that is compatible with astropy.  For example:
```
In [1]: import numpy as np                                                                          

In [2]: from astropy import units as u                                                              

In [3]: unit = u.K * np.sqrt(1 * u.second)                                                          

In [4]: unit                                                                                        
Out[4]: <Quantity 1. K s(1/2)>

In [5]: unit_str = "{}".format(unit)                                                                

In [6]: unit_str                                                                                    
Out[6]: '1.0 K s(1/2)'

In [7]: roundtrip = u.Quantity(unit_str)                                                            

In [8]: roundtrip                                                                                   
Out[8]: <Quantity 1. K s(1/2)>

# Convert to micro K
In [9]: unit.to(u.uK * np.sqrt(1 * u.second))                                                       
Out[9]: <Quantity 1000000. s(1/2) uK>

# Get the actual (unit-less) conversion factor
In [10]: unit.to(u.uK * np.sqrt(1 * u.second)).to_value()                                           
Out[10]: 1000000.0
```

jeffiuliano, 2021-04-16:
I was actually about to say: turns out things like aman.signal are just ndarrays.  It is possible to subclass ndarrays to add an attribute (like ndarray.unit), but I'm pretty sure this is fundamentally what astropy.units does.  So if we wanted to go that direciton, we could just use astropy.

tskisner, 2021-04-16:
Yes, that is what astropy does:

https://docs.astropy.org/en/stable/units/quantity.html#numpy-functions

kmharrington, 2021-04-16:
I agree with Ted that we don't want to turn our whole arrays into Quantities. I've worked with them enough to not want to get stuck with them once we're running lots and lots of data. 

I do like the idea of using them as the indexing part instead of just text/string versions. Because then we can have units on our conversion factors. I'm pretty sure there isn't a 'DAC' unit though, so maybe a "try to use astropy but fall back on strings if they aren't there." I don't expect we'll ever use DAC for more than just the very beginnings of analysis.

tskisner, 2021-04-16:
Note that I have never personally tested performance of using Quantities for large data arrays.  We are trying to use astropy units everywhere in toast now, but mainly for metadata.  We already have have numpy array wrappers around larger blocks of C-allocated memory, so we likely won't try to wrap things as a Quantity.  But for toast I was thinking of just adding an attribute of our DetectorData class that is the unit that applies to the whole thing.

jeffiuliano, 2021-04-16:
You can define your own units, so if we really wanted we could have DAC units.  I would also be inclined not to turn all the data into Quantities -- especially since we'll not be converting it very often, and almost never outside of predefined operaitons (like DAC to Phase to current to power).

I think the aman._units dictionary works fine.  You can then easily see the units of everything important, and update it when you run things like phase to current conversions.

kmharrington, 2021-04-16:
Makes sense.

I think what we need to decide is what to do about the AxisManager version of the `DetectorData` for tracking them.

I'm really leaning  toward getting AxisManagers a `_units` attribute that is similar to the `_assignments` attribute. The assignments  attribute is a dictionary that is tracking which fields are aligned on along which axes. The units one could track which fields have which units. @mhasself  what do you think of this setup? My guess is if we make everything default to something like `None` it won't break any existing code. 

If we do that then I can envision a calibration module that handles unit switching.

mhasself, 2021-04-20:
Sure, sounds fine.  We'll need some new arguments to wrap() and wrap_new() to help units get set automatically, or copied from an existing field.  Also let's not let this spread too far beyond "signal" and calibration information... I do not want to have to start checking whether the boresight az is in radians or degrees!

**But -- if you're already thinking of adding a "DAC" unit... consider:**

There is sort of a conceptual substitution here, where the problem is that we have different _physical properties_ we want to express, and _units_ are seen as the solution to that.  In the lab, of course we say "it's in pA, not pW" ... but that's a shorthand for  "it's the squid current, not the detector power".

If astropy.units permits us to define new units, such as "DAC", then we could also define "SquidPhase_rad", "SquidCurrent_pA", "TesCurrent_pA", "TesPower_pW".  These conceptual distinctions are somewhat more powerful than simple SI unit combos, because we can use them to distinguish between SquidCurrent and TesCurrent (which are both currents, but not necessarily equal).  We could also have "K_RJ" and "K_CMB" and "K_cryo", which I am quite excited about because K_RJ vs. K_CMB is implicated in 110% of all confusion.

Calibration units, such as the output from an IV curve, would be presented with "units" that will cancel properly -- e.g. [TesPower_pW / DAC].

Most of the conversions we do will not be simple "multiply by 1" things ... W -> pW.  They will change the meaning of the signal.  We could be using customized units to express that, and it would be more powerful than simple SI unit combos.

kmharrington, 2021-06-11:
Ok, I'm planning to start looking at this in relation to adding in IV parsing to `G3tSmurf` (going to start making metadata links through Context now that that exists). 

Matthew's point about building in difference versions of units like temperature is a very good one. It looks like astropy does allow both the definition of non-standard units and units that are dimensionless scales of each other (which should be able to deal with things like K_RJ vs K_CMB). The only thing I'm not sure if can do is deal with those scaling when it is not known a priori. But I'll start investigating. 

It's going to require some sotodlib version of a units module, which I'll put in `utils` for now.

kmharrington, 2021-06-11:
Quick update here. The Astropy part of this update looks like it's pretty easy https://github.com/simonsobs/sotodlib/blob/add_units/sotodlib/utils/units.py

Zhilei, 2021-06-11:
We have started writing functions in converting units of intensities, among T_CMB,  thermal dynamic temperature, and W*sr^-1*m^-2*Hz^^-1 
here: https://github.com/simonsobs/sotodlib/blob/4347490cd4f6476aae096390586448d6e39853b1/sotodlib/toast/sim_sso.py#L85
Maybe we can join efforts?

kmharrington, 2021-06-11:
Happy to join efforts, but might also want to point out that a lot of those type conversions already exist: https://docs.astropy.org/en/stable/units/equivalencies.html#thermodynamic-temperature-equivalency

skhrg, 2023-01-23:
@kmharrington are you still working on this? If not I can work on it.

kmharrington, 2023-01-23:
yeah I had what felt like a decently complete sketch of how we'd do it (discussed above) but hadn't really gotten into the implementation. Happy to have you take over!

skhrg, 2023-01-23:
> I'm really leaning toward getting AxisManagers a `_units` attribute that is similar to the `_assignments` attribute. The assignments attribute is a dictionary that is tracking which fields are aligned on along which axes. The units one could track which fields have which units.

Yeah this approach is fairly easy to do and avoids the performance pitfalls of subclassing. I did a quick test and astropy quantities are ~37 times slower at multiplication that a plain ndarray and a simple subclass that doesn't do as much is ~15 times slower than ndarrays (see below). The only potential problems I see with it are:

1. Interface for grabbing units might feel a little clunky since it will be something like `aman._units['signal']`
2. User will have to compute new units themselves if they need them. If you do `aman.signal *= some_value` the units of `aman.signal` won't now be the units of `aman.signal*some_value`.

I don't think either of these are dealbreakers but I figured I would bring them up just in case you had already thought of a solution to them.

For those who are curious about the quick performance test I ran:
```
import numpy as np
import astropy.units as u

class Test(np.ndarray):
    def __new__(cls, input_array, unit=None):
        obj = np.asarray(input_array).view(cls)
        obj.unit = unit
        return obj
    def __array_finalize__(self, obj):
         if obj is None: return
         self.unit = getattr(obj, 'unit', None)
     def __mul__(self, other):
         nu = self.unit * other.unit
         val = self.view(np.ndarray) * other.view(np.ndarray)
         return Test(val, nu)

a = np.arange(100)
b = a*u.m
c = Test(a, u.m)

%timeit a*a
757 ns ± 69.8 ns per loop (mean ± std. dev. of 7 runs, 1000000 loops each)

%timeit b*b
28 µs ± 1.76 µs per loop (mean ± std. dev. of 7 runs, 10000 loops each

%timeit c*c
10.9 µs ± 465 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)

%timeit u.m*u.m
9.02 µs ± 217 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
```

jeffiuliano, 2023-01-23:
You might want to try it on a longer array than 100.  That's short enough that the overhead of setting up the operation might be dominating -- otherwise I'm surprised that it's 15x as long to just do another operation alongside the numpy multiplication.

skhrg, 2023-01-23:
That is interesting, maybe the correct strategy would be to allow the wrapping Quantities then.

kmharrington, 2023-01-23:
Honestly you can probably just wrap quantities now. The question is going to be if things like fourier filters work on them or if wrapping quantities means we'd need to make the whole codebase "unit aware."

skhrg, 2023-01-23:
Yeah, I can run some tests to see what breaks. I suspect that the most things will be fine. Serialization will definitely break but that is fixable.

skhrg, 2023-01-28:
Spent some time poking at it and the only thing doesn't seem to fully work is serializing. Which makes sense since Quantities are subclassed from ndarrays. Serialization isn't even completely broken, it just strips out the units and saves/loads it as an ndarray. Unless there are any objections to using wrapped Quantities to track units I can just update the serialization to handle Quantities.