simonsobs/sotodlib #120: Reconcile G3tSmurf -- Expand Context to include optional Channels table

kmharrington, 2021-04-16:
The overall thought here is that I would like for `G3tSmurf` to work more completely with Context / sotodlib fairly soon, because I am worried about "building too far away" from the main direction. Particularly as there's multiple versions of work heading toward building metadata from SMuRF work. 

I've gone through and verified that, with one minor tweak, the databases built by G3tSmurf can go directly into `ObsDb` and `ObsFileDb`. The piece that is harder is Channels vs Dets. 

I could technically rename some things and turn Channels into Dets, but that seems like a bad idea long term since channels are inherently not detectors. It's also obvious that SO wanting to do Channel-level analysis is going to persist when we're on-sky.

Looking through the code, I think its fairly possible to add some `or`s to the Context code to deal with having either Channels or Dets. Dealing with the combination (when Dets map to a set of channels) will be more subtle, but I think if we always build assuming one detector will map to many channels then we'll be able to deal with that. 

Looking to see if anyone is totally opposed to this idea or has thoughts about where/how this should be implemented.

mhasself, 2021-04-20:
Trying to summarize a discussion we had offline (though I inevitably added some stuff, sorry):

- The single database that G3tSmurf uses can be passed for ObsDb and ObsFileDb -- great!
- There is nomenclature conflict, in that a distinction needs to be made between readout "channels" and the specific physical "detectors" or pseudo-detectors to which they correspond (a thing that is not easy to determine).  Although G3tSmurf is careful about this terminology, it is more annoying to maintain this terminology distinction in Context / AxisManager.
- The Context DetDb can't be mapped into some table(s) in the G3tSmurf database, because DetDb will try to look at all the tables in the database.  Therefore it makes sense for the archive update code to simply write the DetDb information into a separate database file.  The information copied out can be some useful but incomplete subset of the full web of tuning info that G3tSmurf tracks.  This has a nice side-effect of not cluttering up G3tSmurf with unwelcome "dets" terminology!
- The channel -> detector mapping is a complicated map that probably should not live in G3tSmurf database.  If we load it as metadata, you can easily look up, in an AxisManager indexed by Channel names, what the corresponding "detector" name (and other info) is.

So I believe our conclusions were:
- Write out a separate "DetDb" for context use
- Frown but accept that the axis will be called ".dets" even if it indexes "Channels".
- Far downstream, people might want the .dets names to actually be the "Detector" names and not the Channels.  We will not worry about that yet.

One other thing -- the Channel -> Detector map, when treated as metadata, isn't directly supported by default metadata system, because it expects that basically there's a distinct big table of numbers you want to load for each observation or time interval or whatever.  But with the right "loader" function, you could access some Channel Map database, and pull out just the set of Detector description rows that are relevant for a given observation, and put that info into the AxisManager with a 'dets' axis (which, still in this example, has .dets.vals = [_channel_ names]).

kmharrington, 2021-06-11:
Closed with https://github.com/simonsobs/sotodlib/pull/131

closed 2021-06-11 by kmharrington