simonsobs/sotodlib #121: DetDb assumes base table exists

kmharrington, 2021-04-22:
(@mhasself  I can fix this in the update I'm working on for #120 would you like me to?)

DetDb currently assumes that a table called 'base' exists here:
https://github.com/simonsobs/sotodlib/blob/9504613da63b194971ed7a15d33c8064b77fcda1/sotodlib/core/metadata/detdb.py#L404

Could we instead return some sort of "malformed request" error? I encountered this while not using the correct format for the props dictionary and I think that's the most likely user error there.

mhasself, 2021-04-22:
Well, "base" is the main properties table.  That's not documented properly, I'll grant you that.  I'd prefer that the documentation / error messages / validation of DetDbs be improved rather than removing that aspect of the schema.

The base table allows you to access some key properties without needing to state the table name (or hopefully it does... that was the point of having a special one).

kmharrington, 2021-04-22:
Right, I guess I could have put things into `base`. I instantly just put all the channel properties into a `smurf` table because I had to choose a name with `create_table`.

closed 2021-04-22 by kmharrington