simonsobs/sotodlib #125: Window Function for filters

Zhilei, 2021-05-12:
I was wondering whether there is a plan to develop a window function before filtering.

Without the windowing/tapering on the two ends, the Fourier filters generate large oscillations on the two ends.

We can easily cut the ends for sure, but I think applying window functions beforehand may be a more element approach.

kmharrington, 2021-06-09:
So the thing that bothers me about this is you're changing the ends anyway. When you apply a window in the timestream you effectively zero out the ends. Since there's finite precision on these numbers: multiplying by the window, filtering, dividing by the window has the effect of raising the noise level on the edges anyway. 

Am I missing something about the correct way to do this?

Zhilei, 2021-06-10:
You are exactly right, so are you saying we should just filter it without tapering and cut out the data on the oscillatory edges?

kmharrington, 2021-06-10:
That's always what I've done. I basically always detrend before filtering, (which, yes, is technically a window) and then the oscillations can be constrained to the cutoff level of your filter. 

I'm interested if there's advantages to windowing vs. not in the case, but I've never liked the trade off.

kmharrington, 2021-06-11:
Generally curious if this is something that would be widely used. I'm not totally confident it's good data analysis practice. Let me know if I'm wrong? The code additions here aren't complicated.

closed 2021-11-15 by kmharrington