simonsobs/sotodlib #144: stats Module for tod_ops

kmharrington, 2021-06-21:
Add in things like these

https://github.com/simonsobs/LATR_Validation/blob/6d89e3eae8e92c32aaa69218587d54982bfbda1d/latrtools/latrtools/__init__.py#L400-L654

jeffiuliano, 2021-06-22:
Initial commit with some of these functions has been pushed to branch ae00232.  What else did we want to include?  Is this the right place for a noise model and tool to fit psds against it?  (Which can be some combination of the one I modified from pysmurf, the one Katie made, and whoever else has thoughts/needs)?

kmharrington, 2021-06-24:
I think this is the right place to put these. The first two functions will need to be "AxisManager-ified."

It would be nice to have a white noise calculator + 1/f fitting setup in sotodlib. @RemingtonGerras  was starting the white noise side, perhaps you all could bring them together.

skhrg, 2021-08-06:
@kmharrington I've been tasked by Jeff with finishing this up. What do you mean by "AxisManager-ifing" the first two functions?

kmharrington, 2021-08-06:
On a second look, that's a good question. What are you all using these functions for? 

Also, on the average_to function, it'd be cool to have the option to have a new axes added to the AxisManager that is downstamped to that level. (I have a vision of this sort of function being expanded to include things like cuts).

skhrg, 2021-08-06:
@jeffiuliano probably has a better answer as far as intended use case goes, but they seem most useful as something to pass to a fitting function or something.

Adding an option to add the downsampled axes to AxisManager should be easy enough to do.

jeffiuliano, 2021-08-08:
Yeah, that's right.  Mostly I've used these when, say, fitting a histogram results or something.  It's much more likely to be used for something like that, which may not be in an axis manager or aligned with an AM axis (but I guess could be).  So we can give them the same function(am, timestamps=None) format but it seems unusual to use them that way.

One that we might want to use that way (applying to aman timestamps) is a sine function to fit against  (taking phase and amplitude as inputs).

skhrg, 2021-08-10:
Do we want/need anything more than functions to pass on to an existing fitting routine (`scipy.curvefit`, `scipy.minimize`, etc.)? Or is it worth it to make wrappers that call those functions?