simonsobs/sotodlib #212: G3tSmurf : SmurfStatus Revamp

kmharrington, 2022-01-28:
A couple changes need to be made to SmurfStatus:

(1) we need some sort of integrity checking to say we have found a complete enough status object. Right now we do a bit of that in `load_file` but I think it would be better if it was done at the creation of the status object. 

(2) AxisManagers can accommodate scalars now. This means it's much more straightforward to include the status in loaded smurf AxisManager. (Some things like the readout filter parameters would probably be quite useful to carry around so that later filter functions can deconvolve them). 

(3) Should we just make SmurfStatus an AxisManager?

jlashner, 2022-01-28:
:+1: :+1: :+1:  Yes lets make SmurfStatus into an axis-manager and include it with the tod axis managers

skhrg, 2022-02-14:
Is there any need to include the `num_chans`, `mask`, `mask_inv`, and `freq_map` attributes from SmurfStatus in the AxisManager? Seems to be that the AxisManager produced by `get_channel_info` contains all the useful information from these. I'd like to avoid bloating the AxisManager with redundant information, but if people who use SmurfStatus more regularly than me (I'm assuming @jlashner) need them for some reason I can certainly toss them in.

kmharrington, 2022-02-16:
This is a bit related to the second comment in this issue https://github.com/simonsobs/sotodlib/issues/185

I think we could do this two ways, we could have a list of things that are required to be in the status and check all of them or just make sure we're building the status off a "dump" frame. The dump frame is the specific wiring frame that Jack has set up to include a complete dump of the Rogue Tree (I think I'm remembering that right).  Example of us finding the dump frame in the file is here: https://github.com/simonsobs/sotodlib/blob/9bd2bb6a9e9b26d51ab23e5df865d7a7544d9adc/sotodlib/io/load_smurf.py#L1299 

Right now the only thing I've been using from the SmurfStatus in the AxisManager is the channel info building (I use other parts while building the database). In the future we'll want that readout deconvolution and we might end up wanting even more.

skhrg, 2022-02-17:
It's easy enough to include some combination of both I guess. For the readout convolution there is already the `filter_enabled` bool in the status that can be used so all I need to add in the check for the "dump" frame.

PR #226 (2022-03-09): SmurfStatus Revamp
Resolves #212
closed 2022-03-09 by skhrg