simonsobs/sotodlib #215: LabelAxis should only accept strings for labels

mhasself, 2022-01-31:
Be it resolved that LabelAxis should only accept string labels.  The use of integers (and floats, complexes, bools...) as labels is currently supported, to varying degrees, but should instead raise on error on instantiation.

skhrg, 2022-02-10:
Wouldn't it be preferable to have it just cast the labels to strings upon instantiation? At least in terms of reducing how much other code (like #216) needs to be changed.

mhasself, 2022-02-10:
> Wouldn't it be preferable to have it just cast the labels to strings upon instantiation? At least in terms of reducing how much other code (like #216) needs to be changed.

Sometimes, you want to change or improve or extend something without breaking everyone's working code.  But in this case it is not that much code that needs to be changed.  And the main reason for this restriction is that integer labels can cause confusion (meaning code that is difficult to understand) -- so I _want_ to force that change in downstream code.

PR #222 (2022-02-15): LabelAxis should only accept string labels
Resolves #215 and #216
closed 2022-02-15 by skhrg