simonsobs/sotodlib #219: G3tSmurf: Make bias group maps easily accessible

kmharrington, 2022-02-03:
Now that we have more official / complete `sodetlib` bias group maps coming out we should take our version in `latrt-testing` and add it into sotodlib so that it's easily accessible. Eventually it will run through caldbs / the metadata system but for now lets make something that can be automatically loaded and used. The backend loading can get updated once we have a better detdb / caldb setup for g3tsmurf.

kmharrington, 2022-08-26:
Question for @jseibert575 @skhrg @jeffiuliano @jlashner 

With Calibration operations now in the G3tSmurf databases and the ability to get paths of all the files associated those observations [here](https://github.com/simonsobs/sotodlib/blob/master/sotodlib/io/g3tsmurf_utils.py#L31). Do we consider this enough to make bias group maps "easily accessible?"

skhrg, 2022-08-26:
This is good with me. Historically I've been globbing for the path based on a timestamp so this is an improvement for sure.

closed 2023-04-21 by kmharrington