simonsobs/sotodlib #224: G3tSMuRF : Observation updating strategy

kmharrington, 2022-02-11:
We just recently hit two different issues with trying to find the end of an Observation. 

Issue one is more straightforward, I've been using updated to the `'AMCc.SmurfProcessor.FileWriter.IsOpen'` key to decide if an observation ends but it could be that one or `'AMCc.SmurfProcessor.SOStream.open_g3stream'`. So we need to also watch for that one. 

Issue two might be more complicated I always just look at the last file available in a g3session to check for the stream ending information. But we hit an edge case (in `/mnt/so1/data/chicago-latrt/timestreams/16445/ufm_cv4/1644597953_00X.g3`) where there is one single scan frame in the last file and all the stream ending frames are in the second to last file. I'm assuming this is something that just "can happen" with how our g3 files are made so we're going to need to be able to deal with it.

jlashner, 2022-02-11:
Yeah, stream ending in general is a bit tricky. After the end_stream flag is set using either of the two registers mentioned above, the stream will be held open until scan frames stop (and are replaced with idle frames) so that we are sure we've saved any queued / bufferred data.

I think technically the safest way to find the end of an observation is by filename, all files with a given session-id will always correspond to the same observation. If you'd like we can modify the g3 file-writer to add an EndObservation frame or something when a file is closed due to the end of an observation.

kmharrington, 2022-02-11:
oh that last option sounds the best. I moved away from just looking at files because it's tricky to do correctly when the database is updating while data is still coming in. So, I've been leaving `obs.stop = None` until I get an indication that the observation is complete. I think explicitly ending the whole file is my favorite option.

jlashner, 2022-02-11:
Cool, I feel like this should be an `Observation` frame but I think that would mess up some of the other G3tSmurf archiving functions.

kmharrington, 2022-02-11:
I think it might mean we need to change how we handle `Observation` frames in two places but I don't think that means this shouldn't be an `Observation` frame. If this update increases the version number of the .g3 files types then it should be pretty straightforward to work it out?

kmharrington, 2022-02-26:
I've just noticed that at least some of our dump frames have: 
```
'AMCc.SmurfProcessor.FileWriter.IsOpen': False
'AMCc.SmurfProcessor.SOStream.open_g3stream': 1
```

What's the difference between these two?

PR #292 (2022-08-23): Updates to get things working for End-to-end Testing
Various things we needed to update just in general or for end-to-end tests.

* use observation frames to determine if an observation. Technically completely closes #224 even though I apparently decided to close the issue before doing this. 
* add stream_id to the SmurfStatus object because we need a way to pass that around and that makes the most sense
* Add the `no_signal=True/False` option to the g3tsmurf loader. closes #269  even though it's not actually needed for this I just felt like finally doing that.
closed 2022-08-09 by kmharrington