simonsobs/sotodlib #234: ChanAssignments in G3tSmurf Table only including band 7

jseibert575, 2022-03-09:
For some reason, the most recent channel assignment in the tuneset we're currently using is only including the band 7 channel assignments. I think I've figured out it comes from these lines (https://github.com/simonsobs/sotodlib/blob/ac3545d5f3d73e1d5f665e039aaf9dad25c265b5/sotodlib/io/load_smurf.py#L843) in the updating lines to add to the ChanAssignments table, but I'm not sure if this has been a problem for anyone else with other indexes, nor why we haven't seen this before. I'm going to write a workaround but don't want to implement any bugfix until I figure out if this is just an issue with the way we're updating our index now, since this is the first (and only) tuneset copied over since the SupRSync bug was solved that wasn't copying tunesets, so it might be related.

 @kmharrington probably would have seen this the most, so tagging here.

kmharrington, 2022-03-09:
Oh I know why that line of code is in there but don't know why it's now causing you issues. I put it in there because pysmurf started making two sets of channel assignments every time we ran setup notches. It had something to do with trying to speed up the tuning. But, I didn't want to make 2x2000 channels when it was a pretty good bet that 2000 of them would never be used. 

I'm guessing the folder setup has changed to put more than just one band's channels assignments in a folder? 

A quick bug fix is to just delete the line and accept lots of extra channels. A longer elegant fix might be to combing the tune+channel assignment indexing so we only index channels assignments that are called out in a tune file.

jseibert575, 2022-03-09:
Can we check later ctime per band? Like check the band name in the file name, which should generate two files per band, and then just take the later of the two?

kmharrington, 2022-03-09:
yes that's also an option. Will have to add more logic to that function though because right now it assumes that there is only one meaningful channel assignment per folder.