simonsobs/sotodlib #236: When loading data with load_smurf, cannot run `aman.restrict` on returned axis manager

jseibert575, 2022-03-14:
When loading with sotodlib.io.load_smurf, either using `load_data` or `load_file`, the returned axis manager cannot successfully run `aman.restrict`. This causes a crash when using `load_data`, [here](https://github.com/simonsobs/sotodlib/blob/d33e5224e0721bdc69c4af9f1fd22961bff21550/sotodlib/io/load_smurf.py#L1256), but the same error persists when data is successfully loaded from a g3 file, and then I try to run the `aman.restrict` line manually. 

The bug seems to be directly related to the `aman['status']` sub axis manager, when the code in `aman.restrict` tries to run [this line](https://github.com/simonsobs/sotodlib/blob/d33e5224e0721bdc69c4af9f1fd22961bff21550/sotodlib/core/axisman.py#L721). Because the thing stored in `aman['status']` is now an axis manager, it tries to run `aman['status'].copy()` which gets to [this line](https://github.com/simonsobs/sotodlib/blob/d33e5224e0721bdc69c4af9f1fd22961bff21550/sotodlib/core/axisman.py#L288), and fails because the first key/value pair in `aman['status'] is `dump_frame` and then either a `bool` or a `None`, which has no method `.copy()`. 

I'm not sure if this is a bug in the way the new SmurfStatus object/axis manager is created (@skhrg ) or something that axis managers are more generally expecting (@mhasself ), i.e. that the logic in the way axis managers are constructed is incorrectly assuming something about the structure of any axis managers passed to `aman.restrict`, so I'm not sure what the correct path to fix this is, but I think it should definitely be fixed somewhere. 

The error can be reproduced by running the following code, if one is so inclined:

`import sotodlib.io.load_smurf as ls

aman = ls.load_file('/mnt/so1/data/ucsd-k2so/timestreams/16456/ufm_mv11/1645644498_004.g3')
aman['status'].copy()

`

OR:

`
import sotodlib.io.load_smurf as ls
aman = ls.load_file('/mnt/so1/data/ucsd-k2so/timestreams/16456/ufm_mv11/1645644498_004.g3')
aman.restrict("samps", (0, 120014))
`

Let me know if I can give any more details, or answer any questions.

skhrg, 2022-03-14:
Ah, I think what's happening here is that the `dump_frame` field in `aman.status` is being stored as a scalar in the AxisManager and the scalars don't necessarily have a `copy()` method since they are just bools or ints or whatever. I'll add a check in the `copy()` function to handle scalars properly.

skhrg, 2022-03-14:
#237 fixes this for me, if someone could review and approve it that would be great.

jseibert575, 2022-03-14:
I'd like @mhasself to take a look to make sure that editing the axis_manager code is the correct approach, instead of fixing the construction of the status axis manager, but otherwise thanks! If he is okay with that I can review and approve.

skhrg, 2022-03-15:
This issue generically effects any AxisManager with a scalar (I really should have caught this when I was adding scalars to AxisManagers). So the only way to fix this in the construction of the AxisManager is to wrap the scalars in an object with a `copy()` function or to not store things as scalars at all, both of which add needless overhead.

PR #237 (2022-03-15): Handle scalars properly when making copy of AxisManager
Fixes #236
closed 2022-03-15 by skhrg