simonsobs/sotodlib #249: timezone dependency

KenjiKiuchi, 2022-04-06:
load_data function in sotodlib.io.load_smurf.G3tSmurf class require start time and stop time.
However, these timestamps appear to be dependent on the time zone settings of the analysis machine and the daq machine.
Could you modify the code to be timezone independent?

skhrg, 2022-04-06:
Is the correct thing to do here to just have everything in UTC? At least on the sotodlib end I think that's just a matter of setting `tzinfo=timezone.utc` for the datetimes (probably in `_make_datetime`). Right now when you do `start.timestamp()` it will use your systems timezone unless you set `tzinfo` to something else.

KenjiKiuchi, 2022-04-06:
Yes, I think so.
We confirmed that the sotodlib record the data with UTC.
I think unified to the UTC is a good solution.
Thanks.

skhrg, 2022-04-06:
I'll give it a bit to see if someone has objections or if there's something I'm missing but otherwise I can get a PR in for this in today.

mhasself, 2022-04-06:
Yes, please enforce UTC globally.  Also please mention this in a few of the key docstrings (I'm not super familiar with the code ... but wherever a user would probably first become confused!)

PR #250 (2022-04-11): Set timezone to UTC
Should close #249 
Note that here all its doing is setting the timezone to UTC without doing any conversions. I can instead have it convert the input datetimes to UTC if that is preferred.
closed 2022-04-11 by skhrg