simonsobs/sotodlib #283: Backwards compatibility?

jeffiuliano, 2022-08-02:
Do we expect the observation indexing to work on old data?  Currently indexing is generating errors of the form:

```
INFO: sotodlib.io.load_smurf: Unexplained Error at crate1slot2, 1617483593, /mnt/so1/data/penn-latr/smurf/16174/crate1slot2/1617483593_take_noise_psd
```

on most of our older detector data, and no corresponding observations are being generated.  Maybe there is no good way to make it work, but it might be useful to be able to go back and find old observations if it is doable.  (I suppose we could make a separate db with an older version of sotodlib and just leave it as a reference).

Would be good to be able to suppress the error messages on this older data.

kmharrington, 2022-08-02:
There is limited backward capability but not 100% coverage. A lot of the lack of backward combability comes from where we've made significant changes to the format of the g3 files and have chosen to not keep massive chunks of code around for something that hasn't been fielded. For me, that thought process is going to stay like that. We'll keep things backward compatible where reasonable but not prevent us from making things better where necessary.

That data is from April 2021, there have been a massive amount of changes since then, particularly to what is inside the status frames. We depend on those for tracking readout_id metadata and making sure the correct files are associated with the correct observations. If the required information isn't in the status frames I don't want us making observations. 

Are you getting the same sort of errors from things after ~October 2021?

(side note: I know that is the most useless error message for debugging things, I do have designs on fixing it, #206 ).

kmharrington, 2022-08-16:
I'm going to go ahead and close this since I think it's resolved.

closed 2022-08-16 by kmharrington