simonsobs/sotodlib #297: G3tSmurf need Delete Capabilities

kmharrington, 2022-08-26:
When we get to the field, Level 2 data will be temporary and once data has been bookbound and successfully archived we won't be keeping the level 2 data around. The G3tSmurf instances running at the site will need to adapt to that and remove entries after they are no longer being used. 

This is going to need to work in concert with whatever entity is choosing to delete level 2 data.

jeffiuliano, 2022-08-28:
Two thoughts about this:
1. It will likely be useful for systems debugging purposes to keep as much houskeeping data as possible in certain L2 locations (and it's so much smaller than det data, that we could potentially keep hk data indefinitely).
2. Would it be useful to keep at least some db info when we move the files.  For example, it might be interesting to be able to query all Observations and Files, so you can see what _had_ been there before.

As an example of something one could do with that -- imagine having a cumulative season observation efficiency estimate plot based on the Observations db or something as part of the site status/daily analysis info.  Could also be useful in debugging in case there is ever uncertainty about what happened with certain files.

After files are deleted, Files.name could be empty, or there could be another column recording that the file has been deleted.  We could delete rows in tables like Frames if they would get unwieldy over time, but keep the less-large entries that might be of use.

kmharrington, 2022-08-29:
There will be a whole other set of indexing / tracking of the data at level 3. The observation and file tables are a required part of the context loading system, so as we repackage the data up to level three we'll be transferring that information with them. The whole process of creating books and shipping them to NA is tracked, so we should never be removing things before they're safely stored in their final formats. 

The housekeeping data will also be bookbound and shipped off the more permanent locations. To my knowledge we've never planned to keep any site L2 data around long term, just L3. (lab testing data is different for obvious reasons).

closed 2023-07-27 by kmharrington