simonsobs/sotodlib #302: Batched loading of TOD loads more files than needed

skhrg, 2022-09-13:
When using the new batched TOC generator both @jeffiuliano and I ran across the issue where when loading the n'th tod it seeems to be loading n files. The signal in the returned TOD is still correct but this makes the data loading take increasingly longer at each iteration.

Adding a print statement to `load_file` shows `flist` grow each iteration with the first element being correct and the rest being other files with some odd sample bounds.

Here is an example:

```
In [148]: amans = gu.get_batch(obs.obs_id, SMURF, n_samps=10*60*200)

In [149]: for aman in amans:
     ...:     print(aman.signal.size)
     ...: 
     ...: 
[('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_000.g3', 0, 120000)]
100%|████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:16<00:00, 16.99s/it]
189600000
[('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_000.g3', 120000, 240000), ('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_001.g3', 0, 119966)]
100%|████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:33<00:00, 16.63s/it]
189600000
[('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_000.g3', 240000, 360000), ('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_001.g3', 119966, 239966), ('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_002.g3', 0, 119930)]
100%|████████████████████████████████████████████████████████████████████████████████████████████| 3/3 [00:58<00:00, 19.39s/it]
189600000
[('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_000.g3', 360000, 480000), ('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_001.g3', 239966, 359966), ('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_002.g3', 119930, 239930), ('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_003.g3', 0, 119892)]
 50%|██████████████████████████████████████████████                                              | 2/4 [00:32<00:32, 16.07s/it]
```

Where the `obs` is `ufm_mv15_1662875703: 2022-09-11 05:55:01.989792 -> 2022-09-11 13:37:19.964351 [7:42:17.974559] (None)`

skhrg, 2022-09-13:
Actually seems to happen whenever I specify `samps` to `load_file`

```
In [151]: ls.load_file(sorted([f.name for f in obs.files]), archive=SMURF, samples=(120000, 240000))
[('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_000.g3', 120000, 240000), ('/mnt/so1/data/penn-latr/timestreams/16628/ufm_mv15/1662875703_001.g3', 0, 119966)]
100%|████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:45<00:00, 22.93s/it]
Out[151]: AxisManager(timestamps[samps], status*[filter_pars], det_info*[dets], iir_params*, signal[dets,samps], primary*[samps], biases[bias_lines,samps], flags*[dets,samps], dets:LabelAxis(1580), samps:OffsetAxis(120000), filter_pars:IndexAxis(16), bias_lines:LabelAxis(16))
```

So maybe this is just me misunderstanding some `G3tSmurf` stuff?

jeffiuliano, 2022-09-14:
I can confirm that this is working for me.

PR #303 (2022-09-14): Check if end of file is before requested sample range
Fixes #302
closed 2022-09-14 by skhrg