simonsobs/sotodlib #306: HWPSS Template Subtraction w/ Arbitrary Harmonics

msilvafe, 2022-09-19:
The current implementation of the HWPSS template generation function for template subtraction in the hwp branch ([link here](https://github.com/simonsobs/sotodlib/blob/4aa26dc6b132b484300a95e9326a90abf3d80c29/sotodlib/hwp/hwp.py#L10-L78)) specifies certain "modes" each one corresponding to a different fitting function with a different number of harmonics of the hwp rotation frequency in it. I think this function could be generalized to include an arbitrary number of harmonics (for instance [the ABS paper section 4.2.1 claims](https://arxiv.org/pdf/1801.01218.pdf) that they use the first 20 terms including the mean!!!). One way to do this which appears to estimate the Fourier coefficients just as well is to integrate against the particular harmonic like this:

```
def extract_hwpss_int(aman, signal='signal', name='hwpss_ext',
        hwp_angle=None, modes=[2,4,6,8]):
    """
    Calculates the sin and cos coefficients for a Fourier series only containing the specified harmonics in the expansion for HWP synchronous signal estimation. Nothing is returned by the function, a new axis is added to the input AxisManager.
    Args:
    aman (AxisManager): Axis manager containing data with a synchronous signal in it
    signal (str): Specifies the axis in the axis manager to calculate the template from
    name (str): Name of the new axis to write the template to
    hwp_angle (ndarray): Array size 1 x N_samps that contains the reconstructed angle of the hwp. 
                         If ``None`` then ``aman.hwp_angle``
    modes (int list): List of the harmonics of the Fourier series to estimate amplitudes for. 
    """
    if hwp_angle == None:
        hwp_angle = aman.hwp_angle
    if name in aman.keys():
        aman.move(name, None)
    aman.wrap_new(name, ('dets', 'samps'))
    As = np.asarray([np.sum(aman[signal]*np.sin(m*hwp_angle)*np.diff(aman.timestamps).mean(), axis = 1)\
          /(np.ptp(aman.timestamps)/2) for m in modes])
    Bs = np.asarray([np.sum(aman[signal]*np.cos(m*hwp_angle)*np.diff(aman.timestamps).mean(), axis = 1)\
          /(np.ptp(aman.timestamps)/2)  for m in modes])
    for i in range(aman.dets.count):
        y = np.zeros(len(aman.timestamps))
        for a, b, m in zip(As[:,i], Bs[:,i], modes):
            y += a*np.sin(m*aman.hwp_angle) + b*np.cos(m*aman.hwp_angle)
        aman[name][i] = y
``` 

This takes about the same amount of time to run as the fitting without having to bin the data in hwp angle bins (integration takes ~70 sec and fitting took ~45 sec with 36001 bins for the default here which is fitting 2f, 4f, 6f, and 8f but I think the integration should be able to be sped up by truncating the amount of time you integrate and getting rid of loops in favor of numpy functions and list comprehensions). I think it's also just a little more flexible so that you don't have to write a new function every time you want to explore fitting one more or less harmonic.

yukisakurai, 2022-09-22:
Thank you for your suggestions. I think your point is almost equivalent to the functions implemented in SA and ABS. I have implemented the simplest and stupid method as the first version. We need to include odd terms such as 1,3,5f etc. in real data so this implementation is essential. I will push next version reflecting your suggestions soon and let you know.

keskitalo, 2022-10-18:
Can either of you explain the relevance of drilling down on the spectral composition of HWPSS? If we bandpass-filter and demodulate the data, only power around 4f survives and shows up as a constant offset in the demodulated data.

msilvafe, 2022-10-18:
I think there are two reasons we want to estimate the template:
1. To track the relative amplitude of the HWPSS template components in time since they trace gain variation and different systematics.
2. By removing the template we can then flag and fill smaller glitches that aren't that significant relative to the HWPSS. We plan to do an iterative process where we glitch flag, fit HWPSS template with the flags masked, remove the template, flag glitches again, add back in the template and fit again with the newer set of flags, and potentially repeat that process a few times to catch all glitch flags and get a good estimate of the template (to keep for the reason described in 1) unbiased by the glitches. 

This is at least my understanding of it.

msilvafe, 2023-05-25:
Addressed in PR [#437](https://github.com/simonsobs/sotodlib/pull/437)

closed 2023-05-25 by msilvafe