simonsobs/sotodlib #346: make_det_info_wafer and make_read_det_match failing to match up with add_det_info() in g3tsmurf_utils

davidvng, 2022-11-30:
Attempting to create databases using `make_det_info_wafer.py` and `make_read_det_match.py` and comparing the resulting wafer info to the output of `add_det_info()` in `g3tsmurf_utils`. In an example case, I'm finding that the first method is cutting channels due to not finding matching readout IDs to detector IDs. (1698 IDs as opposed to 1808 IDs using `add_det_info()`). I tried to add a "NO_MATCH" det_id entry to `make_det_info_wafer.py` but that only adds a single ID after loading in using context.

Corresponding PR to all of this: #345 

@kmharrington and I are hoping @mhasself could comment on this?

kmharrington, 2022-12-01:
A bit more context here (pun totally intended). We found this comparing the AxisManagers loaded straight from files vs the ones loaded through Context. The context load pattern follows the one in the docs here: https://sotodlib.readthedocs.io/en/latest/site_pipeline.html#context-file-format 

Right now, the `readout_2_detector_manifest.db` assigns a `det_id` of `"NO_MATCH"` to any readout_id where DetMap doesn't find a corresponding detector.  If we just use the the first manifest db, 
```
metadata:
    - db: "/path/to/readout_2_detector_manifest.db"
      det_info: true
      det_key: "dets:readout_id"
```
this gets us an AxisManager with all 1808 channels present. But 110 of those channels have `aman.det_info.det_id == "NO_MATCH"`. 

When we add the wafer info back:
```
metadata:
    - db: "/path/to/readout_2_detector_manifest.db"
      det_info: true
      det_key: "dets:readout_id"
    - db: "/path/to/det_info/wafer/manifest.db"
      det_info: true
      det_key: "dets:det_id"
```

The loader quietly enforces the constraint that each `det_id` in the AxisManager has an entry in the wafer manifest _and_ that that entry is unique. So, if the wafer manifest doesn't have any entry for "NO_MATCH" then the AxisManager is just returned with 1698 channels and there's no notice that we're cutting some just because of lack of entries. If we have a single "NO_MATCH" entry in the wafer manifest, we get 1699 channels back where one of those "NO_MATCH"s were kept. 

At a minimum, I don't think we want the databases quietly doing data selection for us, there should be some feedback about what's happening. But there's an additional question of if it's possible and/or reasonable to remove that uniqueness constraint as well.

mhasself, 2023-01-05:
I believe this has been addressed suitably by #350 ; closing.

closed 2023-01-05 by mhasself