simonsobs/sotodlib #348: Is update-hwp-angle robust to file boundaries?

kmharrington, 2022-12-04:
Came up in the site data flow call this week. We're currently running `update-hwp-angle` on individual complete files for site automation purposes. 

In this setup, do the output saved files have edge effects at the file boundaries? For example, is there weird data at the beginning of a new file before the HWP has gone over an index? 

If the last sample of the previous file neatly lines up with the first sample of the new file then we're good. If not, we need to update `update-hwp-angle` to deal with edge effects.

yukisakurai, 2022-12-08:
Thank you for sharing. We will investigate it and report back soon (within a week time scale).

kmharrington, 2023-01-05:
I'm a little bit confused about what you are planning for this additional flag field in this case. Since the file boundaries are totally arbitrary with respect to the hardware, it shouldn't matter if we haven't hit an index inside that file. If we've ever hit an index at any time (and haven't lost it / power ) we should know where the HWP is. If the update-hwp-angles analysis script doesn't account for previous knowledge then that's something we need this script to do as well.

kmharrington, 2023-04-26:
Do we need this issue now? If we are retiring update-hwp-angle and changing to observation specific calculations then there should no longer be issues with file boundaries. 

If we don't need this issue we should close it.

yukisakurai, 2023-04-28:
Thank you for your follow up. I basically think it is no longer issue but we need double check. We would keep it until make-hwp-solution converges, meaning early June.

kmharrington, 2023-04-28:
How many of those can be run off the combination of the OCS feeds and the per-observation manifest dbs? I feel like the combination of those can replace everything. 

Do we have a documented list of all the alerts we're going to run at the site? That would be a good place to make sure we have everything.

closed 2023-10-02 by mhasself