simonsobs/sotodlib #353: Bookbinder needs to look for dropped frames in the counters for smurf data, interpolate them, and flag them somehow

kmharrington, 2022-12-20:


kmharrington, 2023-02-09:
From discussion on the data packaging call, there are several parts to this. First: this is mostly dropped samples on the edge of frames instead of entire frames. The issues at the beginning and end of files are different than samples in the middle. 

For the beginning and end of files/observations. G3tsmurf needs to update the times of fields to reflect the timing system times when available. @guanyilun  Can I double check that the times passed to bookbinder for start/stops originate from `obs.start` and `obs.stop`? 

So, if there are missing samples at the beginning or end of the book it's because the latest `obs.start` in a multi-book observation isn't actually the latest sample time. I think these are because counter time != frame.times. I think the easiest way to fix that would be to use counter time to set `obs.start` and `obs.stop` and I can write a g3tsmurf fix to do that.

Sample drops in the middle are a whole different set of issues. We discussed @jlashner  looking at a preprocessing script that would find out the timestamps a-priori so bookbinder can easily slot in missing samples. Seems like the additional run time there isn't too bad? tbd

guanyilun, 2023-02-09:
> Can I double check that the times passed to bookbinder for start/stops originate from `obs.start` and `obs.stop`?

yes that's right

closed 2023-03-16 by algadoreo