simonsobs/sotodlib #355: update-hwp-angle data quality checking

kmharrington, 2022-12-20:
During P10R1 we have a few times when the hardware connections to the HWP messed up. Things like quad not reading out or not having timing (and I don't understand all the implementation details here). But when those things happened it also messed up the automated scripts calculating HWP angles. At one point the whole system hung indefinitely. 

The angle calculations need to:
1. Identify bad data without hanging or crashing
2. Add a field in the HWP angle calculation results that will be used to classify the level at which the HWP angles are to be trusted. (Details of how this is set up should be written into the [flow_to_librarian and tod_format documents](https://github.com/simonsobs/tod2maps_docs)).
3. Return information about failing angle solutions so that humans can go fix the hardware. Should work with @BrianJKoopman and @guanyilun  to figure out how this returned information can get passed to the site alerts pages.

yukisakurai, 2023-01-05:
starting to work on this topics with hwp branch. It is difficult to make 1 perfectly with future data, but we will try to do it with predictions.

kmharrington, 2023-02-09:
We now have a site pipeline monitor in sotodlib: https://github.com/simonsobs/sotodlib/blob/master/sotodlib/site_pipeline/monitor.py 

As part of QC for the HWP angles we'd like the update-hwp-angle script to seen information about solution statuses to the monitor so we can build alters for when humans should do something with the hardware. So for things like when the IRIG timing is bad or when the solutions are throwing errors. This should look similar to what we're now doing with the update-g3tsmurf-db scripts. ( We can share grafana access instructions on slack).

kmharrington, 2023-04-26:
Commenting here I've changed the title of this from update-hwp-angles to make-hwp-solutions based on our new plan for HWP angle solutions. Some of the issues discussed above still need to happen with the new script. In particular, we need to push updates to the data packaging monitor if the solutions are crashing or returning bad data. That will be an important QA step to make sure something isn't going wrong with the hardware.

kmharrington, 2024-02-20:
@tristpinsm  The remaining parts of this issue might be a useful place for you to work on a non-preprocess QA metric. Maybe @ykyohei  could help with what info we should be looking at / posting to grafana?

tristpinsm, 2024-02-27:
@ykyohei I'm looking at what metrics may be useful to extract from the hwp solutions metadata files to record in Influx/grafana, but I'm not familiar with most of these quantities. Here is what I've gathered so far:

For every observation:
- tempted to include all O(1) numbers, like `primary_encoder, pid_direction, logger, offcenter[2]` but maybe some of these are definitely not useful. (`pid_direction`?)
- total number of flagged samples in the observation
- average calculated rate
- something about the template? I don't know enough about what this represents right now

Let me know if this makes sense or if you have any suggestions. :)

ykyohei, 2024-02-29:
Thanks for looking into this. I agree to monitor these values

- primary_encoder
- version
- logger
- average offcenter
- total number of flagged samples in the observation
- average calculated rate

About the template, it will be good to monitor `np.average(np.abs(template))` .
This value represents how large the template is, how non-uniform the measured hwp encoder slot is.

ykyohei, 2024-03-07:
@tristpinsm I looked at the grafana for this. Looks great, thank you.
https://site.simonsobs.org/grafana/d/b740304e-0c55-42bd-8a8e-6986a63bf291/qa-dev?orgId=1&from=1694102180063&to=1709826980063

I have some requests.

- For the flagged samples, could you also show the absolute number of flagged samples?
- For the HWP template, it looks like a simple average of template, but this value `np.average(np.abs(template))` is more informative for us.
- HWP offcenter is incorrectly labeled. we can monitor offcentering of one direction only
  - X -> average offcenter, Y -> std (error of) offcenter

closed 2025-10-24 by ykyohei