simonsobs/sotodlib #356: make-obs-book needs to include readout_ids

kmharrington, 2022-12-20:
Following the [TOD format document](https://github.com/simonsobs/tod2maps_docs), the names used in the "signal" fields should be the readout_IDs for each channel. 

These IDs are generated and tracked by the g3tsmurf databases and I can help setup the interface on that side. We need to know how bookbinder wants the information passed in.

guanyilun, 2023-01-13:
The readout_ids are incorporated in books by #367.

kmharrington, 2023-01-19:
Looking at some of the books on simons1, it looks like we still don't have readout_ids in the books?

Example test:
```
import so3g
from spt3g import core

book = "/mnt/so1/shared/site-pipeline/data_pkg/books/pton-tsat/obs/16735/obs_1673517465_tsat_1101111/"
for frame in core.G3File(book+"D_obs_crate_so04_slot2_000.g3"):
    if frame.type == core.G3FrameType.Scan:
        print( repr(frame["signal"].names))
        break
```

Looks like the book was last edited today, so I think this is recent.

guanyilun, 2023-01-19:
Just checked this book. It is one example when imprinter failed to parse readout_ids and proceeded. Now it's failing with an error with the bookbinder updates merged.

kmharrington, 2023-01-19:
Ok great. I've checked some from UCSD and see the names are right. 

What's the easiest way for me to get a book and find the level 2 observations they're made from? That way I can figure out why we're missing readout IDs on the tsat side.

kmharrington, 2023-01-19:
Ok, I figured out how to query the imprinter database. The TSAT readout ids are failing because of how they'd set up their tuning (they're running a tune file on one stream id that was made on a different stream id). So all the indexes are performing correctly based on how they've been told to work. TBD if we should allow stream_ids to mix...

But I think books failing when we can't find ids is still the right behavior here. And when we can find the ids that's working. So, reclosing the issue.

closed 2023-01-19 by kmharrington