simonsobs/sotodlib #357: Demonstrate multi-wafer books and smart fall backs

kmharrington, 2022-12-20:
I think this is a combination of `make-book-plans` and `make-obs-books` and I'm not sure where the line is between them. But with P10R1 (and P10R2) data we have examples of SMuRF data that is co-sampled across different UFMs. 

* We need to demonstrate the we can find when books need to be made with multiple wafers in a book and then make them. 
* We also need to have a fall-back structure where if the data are not co-sampled we can bind them into single-wafer books.  (we could test that with LATRt data where we're reading multiple UFMs but they aren't co-sampled).

guanyilun, 2023-01-12:
On the second point, how should the fall-back mechanism work? it doesn't sound very practical to check the timestamps in each frames across multiple observations. Is there any flag that says the data is cosampled so I could just refer to that?

kmharrington, 2023-01-19:
@jlashner  @msilvafe  Is there anything within the .g3 files we can use to know if the timing system is operating correctly / we can assume the timestreams are co-sampled between UFMs? 

It's not insane to check some of the frames in the files to be sure it's working. Right now it definitely isn't with TSAT, they don't have a timing system, ex:
```
book = "/mnt/so1/shared/site-pipeline/data_pkg/books/pton-tsat/obs/16735/obs_1673517465_tsat_1101111/"

for frame2 in core.G3File(book+"D_obs_crate_so04_slot2_000.g3"):
    if frame2.type == core.G3FrameType.Scan:
        break
        
for frame3 in core.G3File(book+"D_obs_crate_so04_slot3_000.g3"):
    if frame3.type == core.G3FrameType.Scan:
        break
        
f2_times = np.array([t.time / core.G3Units.s for t in frame2["signal"].times ])
f3_times = np.array([t.time / core.G3Units.s for t in frame3["signal"].times ])

print( np.median(f2_times - f3_times), np.std(f2_times - f3_times))
```

kmharrington, 2023-01-19:
Oh, other thought: If we can get some flag in the files that tell us it's running off the timing system then we can add a column to the level2 tables to indicate it's timing level. That way imprinter could work off that table.

kmharrington, 2023-01-19:
Ok, I am going to update my previous comment and say we HAVE to have something in the files to tell us if there is a timing system running. Comparing books from TSAT (no timing) and UCSD (with timing) I'm seeing that we aren't going to be able to tell by the timestamps alone. Writing some fast math to guess timing difference here is probably doomed to failure:

[image]

But also, I'm hiding another error in the multiwafer books, it looks like we aren't totally lining up the timestamps with the sample numbers. This is the histogram of `f2_times - f3_times` from the earlier code snippet ( the "with timing" is using one of the UCSD books instead). Those |0.002|s offsets are single sample mis-alignments between the different wafers. 

[image]

msilvafe, 2023-01-19:
Can you use this logic: https://github.com/simonsobs/sotodlib/blob/0f3c7b7a0ff882d39cd3e3b8c2258ede0c7ba308/sotodlib/io/load_smurf.py#L2149-L2151

kmharrington, 2023-01-19:
Is this logic too restrictive? Looking at the TSAT data all the counters are all zero. Can we just check for counters that are zero vs counters that are not zero?

msilvafe, 2023-01-19:
Yes, this is the description of the frame header from pysmurf: https://github.com/slaclab/pysmurf/blob/main/README.SmurfPacket.md I think in the case of no timing system the only counter to increment would be the frame counter but counters 0, 1, & 2 would be 0.

guanyilun, 2023-01-20:
Thanks! Now I'm a bit confused. What's the difference between the output of `_get_timestamp` vs just reading `frame['data'].times`?

kmharrington, 2023-01-20:
When the timing system is active, there are counters in the .g3 frame primary data that can be used to calculate the _exact_ time of every data point. `_get_timestamps` is probably not the function we want bookbinder to use, it's really set up for data loading into AxisManagers, not re-writing .g3s. The keys of `streams` are also the keys in the .g3 frames. So I would use that function as a guide for setting it up with bookbinder. 

And, those counters are integers so we should be able to use them to exactly line up the data between different UFMs.

guanyilun, 2023-01-20:
Do you mean that, instead of using frame['data'].times, bookbinder should reimplement this _get_timestamp function for timestamps?

kmharrington, 2023-01-20:
Yeah I think you should use the math in the `_get_timestamps` function to re-write `frame['data'].times` as the correctly co-sampled times. (And we should make sure this is documented in the tod2maps docs)

jlashner, 2023-01-20:
That's right, the timestamps in `frame['data'].times` are the times at which data is first received by the smurf-streamer, so those are not very accurate, and true timestamps need to be reconstructed from the timing counters like is done in the _get_timestamp function. Like Max said, checking if Counter0 is all 0, or if the mean is 0 should be sufficient in telling whether or not there's a timing system.

closed 2023-03-02 by kmharrington