simonsobs/sotodlib #375: Add Timing Column To G3tSmurf Tables

kmharrington, 2023-01-20:
We should be able to see if timing is active or not as we parse through the level2 files. And if we track the column here then imprinter does not need to open the files to figure out how it should bind the data.

kmharrington, 2023-01-30:
@jlashner I'm looking at adding the column to the g3tsmurf databases for the timing system and I'm noticing 2 things about the frames we have in the g3 files from UCSD. 

1. The scan frames have a "timing_paradigm" entry that is a G3 String that is set to "Low Precision" 
2. The "slow_primary" entry in the frames have a "TimingConfiguration"  entry that seems to be set to zero. 

This strikes me as way too much "timing information" in these frames for us to be having to use the values of the counters to find out if the timing is on or off. Can we make these entries actually useful? And if we can't make these actually work, I'd like to have the information removed from the frames since is obviously misleading.

jlashner, 2023-01-30:
Ah yea thanks for reminding me about that, we added `timing_paradigm` a while back but did not update it to work with a timing system.... I can fix this so that it's correct, but the paradigm should be that it's "High Precision" whenever Counter2 is not 0.

The "TimingConfiguration" in the "slow_primary" data comes directly from the smurf header ([here](https://github.com/slaclab/pysmurf/blob/main/README.SmurfPacket.md)), which looks like it should also distinguish between smurf timing modes. However I haven't yet compared this between data with and without a timing system, so I am not sure if it's accurate or not. If it's not accurate we should just remove it

kmharrington, 2023-01-30:
Ok great, I will write the timing column information now using counters, but then could you let me know when these are updated? (and/or link issues here for us to track?)

commit fff2ca39 (2023-01-31): fix #375 and fix #316
commit ddcdfcb5 (2023-02-01): fix #375 and fix #316
commit 3e25f591 (2023-02-07): fix #375 and fix #316
closed 2023-02-07 by guanyilun