simonsobs/sotodlib #388: Book format requests

mhasself, 2023-01-30:
I have some requests for the obs/oper Book format.  I'm looking at, for example, /mnt/so1/shared/site-pipeline/data_pkg/books/ucsd-sat1/oper/16748/oper_1674860955_sat1_0010000 and trying to write Context indexing and loading tools.

1. Info in M_index.yaml:

- "telescope" field (e.g. "sat1")
- "book_type" (e.g. "oper")
- include a list of the included "stream_ids" (e.g. ["ufm_mv14", "obs_ufm_mv6"]).  (These should be in standard order, i.e. in the same order that is used to generate slot-flags.)
- If possible, include the g3tsmurf "detset" name for each stream_id ({"ufm_mv9": "sch_ufm_mv9_1673656362", ...}).
- include a dump of the sample range for each file_index; e.g. if there are files 000 and 001 then "samples_ranges" might be = [[0, 10000], [10000, 12543]].

2. When there are no interesting housekeeping data, the "ancil" entry should still be populated.  It would be simply a G3TimesampleMap with .times populated (it's the .times I want).

3. The "sample_range" (in Scan frames) should store index ranges in Python range notation, where the second index is not included.  For example, the first 10000 samples would be denoted as [0, 10000], not [0, 9999].

4. The filename for D_ prefix files is supposed to be `D_{stream_id}_{file_index}.g3`.  But the books I'm looking at include the book type in there, e.g. `D_`**oper**`_ufm_mv19_000.g3`.  I don't think we need that.

I might tack on a few more things as I keep working on this.

mhasself, 2023-01-30:
5. The "sample_range" should be with respect to an increasing "sample index" that spans all files in an observation.  For example, the file listed below has a first Scan frame with sample range (0, 9999).  But I expected to see (55000, 55999).  (And as per 3 above it should ultimately be (55000, 56000)).

`/mnt/so1/shared/site-pipeline/data_pkg/books/ucsd-sat1/obs/16749/obs_1674963154_sat1_1111101/A_ancil_001.g3`

kmharrington, 2023-01-30:
Just noting that issue overlaps with this issue: https://github.com/simonsobs/sotodlib/issues/359 

But this is a much more complete list than I wrote up.

mhasself, 2023-03-17:
Looking at recent books, I confirm that items 2 and 4 are fixed.  Thanks!

1, 3, 5 seem unchanged.

closed 2023-03-24 by guanyilun