simonsobs/sotodlib #389: Multi-slot bookbinder sample count bugs?

mhasself, 2023-01-30:
Most multi-slot books I've looked at have problems with the sample count -- the stream_ids all have different numbers of samples.  This may simply be due to unpatched dropped frames (#127).  But in at least one case the acquisition is short enough that dropped frames are unlikely, yet still there is a sample count discrepancy.

`/mnt/so1/shared/site-pipeline/data_pkg/books/ucsd-sat1/obs/16747/obs_1674766989_sat1_1111101`
```
ufm_mv14:  N= 11370
ufm_mv18: N= 11370
ufm_mv19: N= 11369
ufm_mv22: N= 11369
ufm_mv6: N= 11369
ufm_mv9: N= 11369

algadoreo, 2023-01-30:
My guess is that the smurf-streamer based timestamps is causing some samples to fall on the wrong side of the book’s end time. Let’s see if they still persist after the timing system issue gets fixed (#387).

mhasself, 2023-01-30:
Understood.

Are the time boundaries set to be squarely _between_ adjacent samples (to minimize risks of rounding error)?

algadoreo, 2023-01-30:
No—the start/end times are pulled by the Imprinter from the database and passed on to Bookbinder. But now that you mention it, I’m not sure how those start/end times are generated.

Nonetheless, if the timing system is on (and everything works correctly), every wafer should have identical timestamps, right? Would it still be sensitive to rounding errors in that case?

algadoreo, 2023-01-31:
@guanyilun pointed out that the culprit could also be the gap checker, since it would add a sample if it thinks the gap between frames is too long (a symptom of the inaccurate smurf-streamer timestamps). Either way, we can see what happens after timing system issue is resolved.

algadoreo, 2023-03-08:
I believe this got fixed by #401—this was the earliest PR I found where the output produces the correct number of samples (N=11369) for all wafers. Although jlashner’s `get_book_times` function mentioned above returns a list with 11370 samples, the last one falls outside the book’s end time and is trimmed.

Perhaps @guanyilun can verify if that PR did anything that changed the behaviour in the right way, e.g., added precise timestamps for start/end times?

mhasself, 2023-04-06:
I think this is all fixed now...

closed 2023-04-06 by mhasself