simonsobs/sotodlib #413: core is not imported before this line is used

jseibert575, 2023-02-19:
https://github.com/simonsobs/sotodlib/blob/1c0807953428455c9239a1ce0246696c97077531/sotodlib/tod_ops/fft_ops.py#L211

I'm happy to fix this myself, it's very straightforward, but wasn't sure if there are proper import things implied that I'm not aware of. If not, as simple as adding the proper import statement to the top of this module. 

Example traceback below:

```---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-21-e3a29a367a11> in <module>
----> 1 f, Pxx = tod_ops.fft_ops.calc_psd(aman,nperseg=len(aman.timestamps)//2,merge=True)

/mnt/so1/shared/software/sotodlib/000000_unstable/lib/python3.7/site-packages/sotodlib/tod_ops/fft_ops.py in calc_psd(aman, signal, timestamps, merge, **kwargs)
    209     freqs, Pxx = welch( signal, 1/np.median(np.diff(timestamps)), **kwargs)
    210     if merge:
--> 211         aman.merge( core.AxisManager(core.OffsetAxis("fsamps", len(freqs))))
    212         aman.wrap("freqs", [(0,"fsamps")])
    213         aman.wrap("Pxx", [(0,"dets"),(1,"fsamps")])

NameError: name 'core' is not defined

mhasself, 2023-02-20:
Go for it, but be aware that the next two lines are also obviously broken.  That code path was clearly never tested.

mhasself, 2023-02-20:
Also looks like this was all fixed in https://github.com/simonsobs/sotodlib/pull/371

kmharrington, 2023-02-28:
Closing this because it should be fixed.

closed 2023-02-28 by kmharrington