simonsobs/sotodlib #443: Detector angle convention in presence of HWP

mhasself, 2023-04-11:
@keskitalo @tskisner @smsimon

In the present TOAST treatment of HWP, the introduction of a HWP causes _rotation_ in Stokes parameter space but does not include the _reflection_ that is also a consequence of HWP optics.

See docstring [here](https://github.com/hpc4cmb/toast/blob/819bb46fd762cad6e979f71a3f9e62d8305a4c9e/src/toast/ops/stokes_weights/stokes_weights.py#L31).  The introduction of the HWP causes the direction of polarization sensitivity to be modified from _a_ to _a + 2w_.  (The implementation seems to agree with the equations in the comment.)  In fact it would be more accurate to say that the sensitivity direction changes from _a_ to _2w - a_.  There is a reflection of the detector's effective orientation, about the HWP's fast axis.

For TOAST / sotodlib, this affects how detector pol angle _a_ is defined, as an input for mapmaking.  Does this angle correspond closely to the actual orientation of the detector in the focal plane, in a sense that would be equally valid if the wafer were being used with a HWP or without [case A]?  Or is it some processed version of the angle (_a_ -> _-a_), that takes into account the presence of the HWP [case B]?

If we pursue case A, then I think it might make sense for TOAST to be modified, so that the HWP projection matrix uses 2w - a instead of 2w + a.  The appeal of this case is that our focal plane products, especially early on, will be easier to construct and check if we don't include the reflection at this level.

If we instead pursue case B, then some attention is needed in the sotodlib instrument model -- for SAT focal planes, all polarization angles should be reflected about some fixed axis, after constructing the full optics tube, but before being used for simulations / reduction.  (Or we add some documentation pointing out that we haven't done that, but the angles are to be interpreted as though we had.)

In the [coord_sys document](https://github.com/simonsobs/tod2maps_docs/tree/master/coord_sys), a long time ago, I recommended including, as part of the focal plane information, the detector angle _gamma_hwp_ to use _when a HWP is present_, in addition to the physical angle that one would use when the HWP is not present (_gamma_).  This has its own hazards, namely that some processing might use _gamma_hwp_ and some might use _-gamma_, leading to inconsistencies in cases where _gamma_hwp != -gamma_.

(This was brought to my attention again recently because @tterasaki noticed some sign convention inconsistencies in analyzing sim data.)

tskisner, 2023-05-11:
Because the HWP simulation code in TOAST has not been revisited for a while (especially since the recent formalization of the focalplane coordinates), I decided to revisit the math.  Here is a document for review, but I will open a separate issue / PR soon in TOAST that adds the missing per-detector gamma offset discussed at the end.  I also see some other double negative signs in the code that cancel each other, and will clean that up too.  I'll plan on adding this doc to the new jupyter-book documentation.
[HWP_algebra.pdf](https://github.com/simonsobs/sotodlib/files/11454853/HWP_algebra.pdf)

tskisner, 2023-05-14:
Ok, I cleaned up some notation in this doc and also added some labels to the figures.  The next step is to open a PR in toast for these changes, which I am working on:  
[HWP_algebra.pdf](https://github.com/simonsobs/sotodlib/files/11472354/HWP_algebra.pdf)

tskisner, 2023-05-17:
This issue in toast is now being tracked here:

https://github.com/hpc4cmb/toast/pull/663

We should keep this current issue open until after that is merged and we have ensured that the detector gamma angle is contained in our focalplane tables generated by sotodlib.

tskisner, 2023-06-22:
Just a note that this is merged upstream:

https://github.com/hpc4cmb/toast/pull/663

With additional unit tests for several HWP state / angles.  Leaving this open until small tweaks to sotodlib are made.

mhasself, 2023-08-15:
Thanks for the fixes @tskisner.

Do any tweaks to sotodlib remain, or can we assume that sims generated through sotodlib wrappers will now have the reflection HWP reflection included?

tskisner, 2023-08-15:
The remaining changes were merged in #505, so I believe we can close this.  However, I do have an action item to go re-check the drone sim workflows.

closed 2025-04-17 by mhasself