simonsobs/sotodlib #462: RangesMatrix doesn't respect AxisManager restrict_axis

guanyilun, 2023-05-09:
When working with RangesMatrix / Ranges in an AxisManager with OffsetAxis, it seems that the indices in Ranges.ranges() refer to indices in the offsetted axis, because [it's built from bitmasks](https://github.com/simonsobs/sotodlib/blob/master/sotodlib/flags.py#L95) that don't know anything about offset. When an axis undergoes `restriction` (axis restriction needs indices in an unoffsetted axis), the indices in a RangesMatrix remain unchanged, so without knowing the exact order of operation I'm unable to figure out what the indices in ranges() refer to. Is this a real problem or am I not using these objects correctly?

mhasself, 2023-05-09:
I understand the gist of your concern here, but could you construct a short example that demonstrates the issue directly?

The "offset" is the property of the axis, and is not normally attached to any items carried by the axis.  Numpy arrays don't store their offsets... so RangesMatrix probably doesn't need to either.

One way to keep track of offset is to carry your RangesMatrix in an AxisManager, with an OffsetAxis...

kmharrington, 2023-05-09:
I remember that [Ranges have references](https://github.com/simonsobs/so3g/blob/master/include/Ranges.h) that I think are supposed to act like an Offset? 

We could just say we always have to have the observation related RangesMatrixes in AxisManagers. I'm not totally sure if that solves all of what Yilun is seeing, so examples could help. But I'm also wondering if we don't want to add get that reference tracking piece of the ranges more accessible.

guanyilun, 2023-05-09:
ah I think it may not be a real problem after all. After more careful attempt, It seems that axis manager is handling ranges  restriction correctly... I think I might have got myself confused with the restriction being done in-place.

mhasself, 2023-05-09:
> I also found that If I start with an offsetted axis,

In this case, your restriction is invalid because your start index (20) is less than the offset of the samps axis (100).  For offset axes, you have to account for the offset when you restrict the axis (see #223 ...).

E.g. this fails:
```
aman = AxisManager(
    OffsetAxis('samps', 100, 100),
    LabelAxis('dets', ['a','b'])
)
aman_sliced = aman.restrict('samps', slice(20, None), in_place=False)
```
But this works and probably gives you what you were expecting the above line to give you:
```
aman_sliced = aman.restrict('samps', slice(120, None), in_place=False)
```

guanyilun, 2023-05-09:
ah ok, thanks! I'm closing it now since my original problem is resolved and the relative offset in restriction is a previously known issue...

closed 2023-05-09 by guanyilun