simonsobs/sotodlib #508: Sim focal plane footprints are weird now

mhasself, 2023-07-21:
@keskitalo I believe #409 did change the default focal plane.  (cc @nad-dch , @kmharrington .)

The control of detailed orientation of subwafers is cool but this isn't how the tube is assembled.

SAT1 focal plane from commit 4d660fc (shows proper interlocking of wafers):

[image]

After https://github.com/simonsobs/sotodlib/pull/409 (commit cf6221f) (shows fun detailed orientations, but incorrect interlocking):

[image]

The LAT is also affected:
[image]

tskisner, 2023-07-21:
Hi @mhasself , I am right now in the middle of a working on a branch with more testing of the focalplane spacing and HWP angle / detector gamma checks.  I am comparing my branch against the layout generated by latest master HEAD (`dc14cd825`).  For SAT1 230GHz I get this layout:
[sat_f230.pdf](https://github.com/simonsobs/sotodlib/files/12130578/sat_f230.pdf)
Which seems to show correct positioning...

tskisner, 2023-07-21:
I will investigate this a bit more in my branch- were your plots made by dumping SAT sims and then loading back in?  Happy to chat on slack if that is more efficient.

mhasself, 2023-07-21:
No, they're based on SOFocalplane 'quat' entries.  I will check a sim.
```
import sotodlib.coords
from so3g.proj import quat
import sotodlib.toast as sotoast
import pylab as pl

def get_wafer_info(telescope, cache_file=None):
    focalplane = sotoast.SOFocalplane(telescope=telescope)
    qs = []
    for k in focalplane.keys():
        qs.append(focalplane[k]['quat'])
    return qs

tel = 'LAT'
x = get_wafer_info(tel)
qs = sotodlib.coords.ScalarLastQuat(x).to_g3()
xi, eta, gamma = quat.decompose_xieta(qs)

pl.scatter(xi, eta, s=2)
pl.savefig(f'fp_{tel}.png')
pl.clf()
```

tskisner, 2023-07-21:
This should be fixed now in #509.  Running your script for both `LAT` and `SAT1` gives the attached plots:
[image]
[image]

PR #509 (2023-07-22): Fix logic when parsing det_info tuple.
This fixes an immediate bug in SOFocalplane.  However, the unit test for this is pending as part of #505.

Closes #508.
closed 2023-07-22 by mhasself