simonsobs/sotodlib #525: Shorthand for common wcs_kernels

mhasself, 2023-08-17:
wcs_kernels are used in sotodlib.coords to represent pixelization schemes (a way of dividing up the sky into a rectangular grid of pixels) without specifying a full map footprint.  It is basically a WCS representing a single pixel map; with the idea that a mapmaking code can continue the pixelization and create a footprint that contains the data (but is pixel-aligned with any other map that did the same continuation).

Although the full space of pixelizations supported by FITS / WCS is quite complex, we mostly deal with one of these:

- For source-centered maps, a TAN projection centered at "ra, dec" = (0, 0), with square pixels (res_x = res_y).
- For sky maps, CAR projection (in the natural orientation), with pixel edges crossing ra, dec = (0, 0), and pixels that are square on the equator.

It would be handy to be able to construct these common kernels  of constructing these kernels by passing in a simple string, e.g.
```
  wcsk = coords.get_wcs_kernel('car-0.5arcmin')
  wcsk = coords.get_wcs_kernel('tan-0.25deg')
```
Moreover it would be nice to be able to pass those strings into functions that want a wcs_kernel ... for one-liners like:
```
  p = coords.P.for_tod(tod, wcs_kernel='car-0.5arcmin')
```

tanaybhandarkar, 2023-09-21:
@mhasself are these the only resolutions we would care about? Or would we potentially want something more general, like the option to pass something like `car-1.0arcmin` too?

mhasself, 2023-09-21:
I think we'd want at least a few of these.  You could start with [1, 1.5, 2, 3, 4, 5, 6, 8, 10, 12, 15, 20, 25, 30, 40, 45] arcsec / arcmin / deg, with deg probably only going up to 10.

Fine with me to just parse the string and extract the resolution; though there's something nice about making it hard for people to have resolutions of, like, 1.23452 arcmins or whatever.

mhasself, 2023-10-10:
Closed via #552 , thanks.

closed 2023-10-10 by mhasself