simonsobs/sotodlib #527: Relax Tag Comparison Constraint + Better Error Logging

kmharrington, 2023-08-17:
https://github.com/simonsobs/sotodlib/blob/6600b79df98041f6b8da924bc5c93e27cb81380c/sotodlib/io/bookbinder.py#L749

This assertion is currently preventing book `obs_1691876375_satp1_0111000` from being bound at this site. This is because the streaming was started by humans (not sorunlib) and it looks like there are differences in capital letters between tags in the different level 2 observations. 

I think we need a way to relax the tag comparison constraint here, maybe it's a flag where we come in and set it by hand to rebuild a failed book? It shouldn't be an issue with sorunlib started observations but I doubt this is the last time we hit this. 

Also, this error did not appear in the `Z_bookbinder_log.txt` file. That file makes it look like book binding completed without errors. We need the bookbinder.get_metadata function to log errors as well.

jlashner, 2023-08-17:
Yeah, I think we can just remove this constraint. I understand the idea that we may want all ufm tags in a book to be the same, but it is easy to mess up. This is definitely being triggered because I mistakenly made the tags different across UFMs, since I started streaming on all of the manually:
```
obs,stream,overnight_hwp_2hz_normal_08/12
obs,stream,overnight_hwp_2hz_normal_08/12
obs,stream,overnight_hwp_2Hz_biased_08/12
```

I think if this is not being caught up-stream to separate different tags into separate books, then we should not enforce it here.

mhasself, 2023-08-17:
> I think we need a way to relax the tag comparison constraint here, maybe it's a flag where we come in and set it by hand to rebuild a failed book? It shouldn't be an issue with sorunlib started observations but I doubt this is the last time we hit this.

I generally like this approach ... request a rebind with, like, "--force-tags= ...." or "--ignore-tag-problems".  This sort of thing is going to happen a lot (minor issues preventing binding), so it's worth having infrastructure in bookbinder to be able to rebind in a (selectively!) more tolerant mode.

closed 2023-09-05 by kmharrington