simonsobs/sotodlib #536: Speed up glitch flags

kmharrington, 2023-08-24:
This line takes a very long time to run: https://github.com/simonsobs/sotodlib/blob/2d6e0467bf2457b0f3a71a84217069c36802c42f/sotodlib/flags.py#L98 

We need to add some logic to look at how long the arrays are and only use a fraction of the array to get these statistics when they are very long.

mhasself, 2023-09-27:
@tanaybhandarkar Here's one.

tanaybhandarkar, 2023-10-02:
I have been playing around with the downsampling option that @msilvafe suggested. Right now, `get_glitch_flags` takes 30-50s to run on a AxisManager of shape (2580, 264000). About 10s is spent on the Fourier filtering, 20-40s on the `stats.iqr` step. To study the effect of downsampling, I just ran the `stats.iqr` step on the calculated `fvec` and looked at the distributions.  I also had to set `merge=False`, so I'm not sure how much time that step takes. 

[image]

Here, "ds" refers to the downsampling factor (i.e., ds= 50 means I used every 50th point: `fvec[::50]`). ds=1 means no downsampling, and `t` is elapsed time. Based on the distribution, ds=2 seems to remain fairly close to the original, non-downsampled distribution, yet only takes half as much time.  

I think setting ds=2 would help a lot here, and I could add a check to downsample when arrays are larger than some number.

mhasself, 2023-10-02:
I want to advocate for a higher ds factor.  Like, 20.  Actually I think you should set the downsampling (or decimation, really) so that the vector you pass in to stats.iqr is small-but-not-too-small.  In this example 264000  / 20 is 13200.  So you could pick ds so that your sampling vector is ~20000 points.

I think a few percent bias or noise in iqr_range is acceptable.  Really what we're trying to compute here is something proportional to the standard deviation (but with outlier rejection).  It shouldn't take many data points to estimate that number.  Think of the stdev of a white noise timestream -- the error goes down as 1/sqrt(Npts)).  You should be able to get a 1% estimate (which is better than we need) using only 10000 points.

mhasself, 2023-10-10:
Addressed in #555

closed 2023-10-10 by mhasself