simonsobs/sotodlib #581: End Sample Glitch Flags

msilvafe, 2023-10-24:
This was flagged (hehe) awhile ago by @bkelle, but it seems like it wasn't fixed. Due to the highpass filtering in `sotodlib.flags.get_glitch_flags` the end samples often get flagged. For awhile, I was just adding in a buffer on the end samples where within that buffer I reset the flags to False (in my own processing not PR'd anywhere). I'm not sure if this is optimal, though since you lose the ability to flag around some number of points at the beginning and end of the TOD. If we think this is ok then I can PR that modification to the existing function. 

Since this largely comes from discontinuities at the beginning/end of the TOD this is helped by running `detrend_tod` first, but in the presence of a big signal (HWPSS, or AzSS for instance) you can still end up with a non-periodic boundary condition, we could consider padding the ends and apodizing to enforce periodicity but this is a bit of a fine-tuning problem (i.e. how much padding/apodizing to apply). We addressed this in the context of HWPSS by implementing iterative glitch flagging between HWPSS subtraction steps we could consider a similar situation with the AzSS subtraction but I'm not sure these two cases span the space of issues that cause these sorts of end flags, and thus would not fully address this issue. Let's discuss here.

Example of TOD (blue) with end sample flags (orange): 
[image]

kmharrington, 2024-01-25:
@msilvafe I think you merged a first-fix for this?

msilvafe, 2024-01-30:
Addressed in: https://github.com/simonsobs/sotodlib/pull/662

closed 2024-01-30 by msilvafe