simonsobs/sotodlib #618: Change book loader to remove context error message?

kmharrington, 2023-11-30:
Even since we started loading with the smurf/detset ManifestDbs we're now getting this error on the AxisManager load: 
```
ERROR: sotodlib.core.context: Key "smurf" is present in det_info returned by observation loader as well as in metadata databases; The two versions are not comparable. dropping the loader version.
```

This really isn't an error, it's that we wanted that smurf key when we had no other information. I think we can just not load those things from the books now?

mhasself, 2023-12-06:
> This really isn't an error, it's that we wanted that smurf key when we had no other information. I think we can just not load those things from the books now?

Sounds good... just so I'm clear, we can remove this (and everything else that make it possible)?

Perhaps we leave the code there, but disable it when using `load_obs_book` but enable it when using `load_book_file`?

kmharrington, 2023-12-06:
I like the idea of disabling when loading from context and enabling when loading from `load_book_file`

mhasself, 2023-12-10:
Resolved in #629

closed 2023-12-10 by mhasself