simonsobs/sotodlib #623: Obsdb Query returns repeated observations

msilvafe, 2023-12-06:
When querying the res_check + take_noise tags like this:
```obs_list = ctx.obsdb.query(tags=['res_check=1', 'take_noise=1'])```
I'm getting lots of repeated obs_ids returned in the obs_list:
```[IN]
for obs in obs_list:
    print(obs['obs_id'])

[OUT]
oper_1698600247_satp1_0000100
oper_1698600247_satp1_0000100
oper_1698600247_satp1_0000100
oper_1698600247_satp1_0000100
oper_1698600249_satp1_0000001
oper_1698600249_satp1_0000001
oper_1698600249_satp1_0000001
oper_1698600249_satp1_0000001
oper_1698600250_satp1_0100000
oper_1698600250_satp1_0100000
oper_1698600250_satp1_0100000
oper_1698600250_satp1_0100000
oper_1698600251_satp1_0001000
oper_1698600251_satp1_0001000
oper_1698600251_satp1_0001000
oper_1698600251_satp1_0001000
oper_1698600264_satp1_1000000
```
Per Matthew's suggestion I also tried querying like this: `query('res_check and take_noise', tags=['res_check', 'take_noise'])` and results were the same.

mhasself, 2023-12-06:
Additional info from quick investigation:

Getting tags for one of these obs from the obsdb shows each tag listed multiple times:
```
>>> ctx.obsdb.get('oper_1698600247_satp1_0000100', tags=True)['tags']
['oper', 'take_noise', 'res_check', 'oper', 'take_noise', 'res_check']
```

The book index.yaml only lists tags once though
```
tags:
- oper
- take_noise
- res_check
```

So that's a clue.

kmharrington, 2024-01-24:
I believe this was all closed by that PR?

closed 2024-01-24 by mhasself