simonsobs/sotodlib #653: Subscan polyfilter too slow

msilvafe, 2024-01-13:
Looping over each detector and then looping over each of the subscans inside the detector loop makes this function quite slow. I think we want to speed it up by finding a more numpy compatible matrix-ified way of doing this bit.

https://github.com/simonsobs/sotodlib/blob/280df0d62efea670868a9ce418ad4d77148d626a/sotodlib/tod_ops/sub_polyf.py#L65-L82

atakeuchi634, 2024-01-23:
I am thinking about filter using Legendre function, which is orthonormal.
Assuming that time stream data is described as TOD(t) = ¥sum_{i=0}^n c_i Legendre(i, t), we can obtain the coefficients of each component by ¥int¥,dt TOD(t) Legendre(i,t).
What do you think of this script? : [sotodlib/tod_ops/temp_mat_polyf.py](url)
(This is temporary!)

mhasself, 2024-01-23:
I haven't used np.ma very much but you can likely reduce object creation overhead in this loop by creating the masked array for each detector only once.  I.e.:
```
for i_det in range(aman.dets.count): 
     if is_matrix: 
         each_det_mask = mask_array[i_det] 
     else: 
         each_det_mask = mask_array 

     # Use these in polyfit instead of new np.ma.array(...)
     t_ma = np.ma(t, mask=each_det_mask)
     signal_ma = np.ma(signal[i_det], mask=each_det_mask)

     ...
```

However, a double loop like this is just not going to be very fast, until you move it into C/C++.  I can help someone write something into so3g for this if that is a priority.