simonsobs/sotodlib #661: Remove WARNING statement from aman loading when `ignore_missing=True`

jlashner, 2024-01-22:
In site-pipeline scripts that build metadata based off of previous metadata, such as the obsdb -> detset -> det_cal -> det_match chain, it makes sense to use a single shared context file with all metadata entries and `ignore_missing=True`, and manually catching cases where the metadata you need is missing, rather than having a separate context file for each combination of required metadata.

This generally works well, except for whenever there is metadata missing (even with `ignore_missing=True`) it logs a warning statement: 
```
WARNING: sotodlib.core.metadata.loader: Failed to load metadata for spec={'db': '/so/home/jlashner/work/daily/2024-01-19/satp3-detmatch/results/det_match.sqlite', 'name': 'det_match'}; ignoring.
AxisManager(obs_info*, det_info*[dets], dets:LabelAxis(1525))
``` 
which clutters up logs. Would it be ok to change this to a debug statement when `ignore_missing=True` or add an additional kwarg to skip this warning?

mhasself, 2024-01-22:
I think I'm ok with that.  I sort of wonder if there's a way to use the `warnings` package, but have those warnings suppressed by default ... I guess that's not much different from burying it in log.debug.

Related: I'm just about done with a rewrite of how metadata are handled in the incomplete / missing cases.  (ignore_missing will still work, but you can also fine-grain target which metadata are ok to ignore.)

jlashner, 2024-01-22:
fine-grained targeting of expected metadata sounds perfect for this, so I can just wait for that!

mhasself, 2025-04-17:
Addressed in #675

closed 2025-04-17 by mhasself